KlipperScreen/KlipperScreen #1240: Graphic interface on the choose printer screen

XJRpilot, 2024-01-15:
### Requested feature

the graphical interface on the choose printer screen is very plain and i would love to be able to choose a graphic of my actual printer similar to the creality sonic pad.

### Is your feature request related to a problem?

no, not related to any problem

thijstriemstra, 2024-01-25:
>  similar to the creality sonic pad.

Can you post a screenshot/photo(s)?

XJRpilot, 2024-01-25:
[image]
[image]

thijstriemstra, 2024-01-25:
Sounds like a nice idea; I'd imagine you'd have to supply the images yourself

XJRpilot, 2024-01-25:
yes that would be great using them. i have no idea how to implement this. you will keep me informed?

closed 2024-04-27 by alfrix