KlipperScreen/KlipperScreen #1241: [FR] Add customizability to "Klipper has shutdown" menu screen

Pastitas, 2024-01-15:
### Requested feature

I'm running a printer in which the power to everything except the rpi is controlled by a relay wired to the rpi gpios. Since the power cuts out after a timeout The machine spends a lot of time in the "klipper has shutdown" screen.
To reconnect one has to go into the setttings>power>printer and turn it on. I'd like to be able to add a button like I can in the main screen with something like this:
```
[menu __splash_screen printer_toggle]
name: Printer Power
icon: light
method: printer.gcode.script
params: {"script":"power_printer"}
enable: {{ 'power_printer' in printer.gcode_macros.list }}
```

alfrix, 2024-01-16:
if you set the power device name equal to the printer name in klipperscreen it would be recognized automatically, if not you need to configure it manually in the printer options (power_devices):
https://klipperscreen.github.io/KlipperScreen/Configuration/#printer-options

you cannot use the macro list because klipper has not started, hence no macros

closed 2024-01-16 by alfrix

mjfsch, 2024-01-18:
It might just be something i was doing wrong, but i tried this i could only get the button to appear on the splash screen if i explicitly specified the power device it in the klipperscreen.conf.

The automatic recognition just would not work with the printer name defined the same as the power device.

Pastitas, 2024-01-19:
I cannot seem to get it to work either, do you have a link to your config @mjfsch?

mjfsch, 2024-01-19:
to make it work, in klipperscreen.conf i added this at the start of the file

`[printer Printer]
power_devices: Printer`

obviously the power device needs to exist in your moonraker.conf

`[power Printer]
type: tplink_smartplug
address: xxxxxxxx
restart_klipper_when_powered: True`

Pastitas, 2024-01-19:
That makes sense but It's not clearly documented, It is working for me now