KlipperScreen/KlipperScreen #1242: [BUG] Pin management. Error

gumileeev, 2024-01-15:
### What happened?

I recently left a [issue](https://github.com/KlipperScreen/KlipperScreen/issues/1237) that the control of the ice pin, which should either be turned on or off, is displayed on the display with a slider.
my switch code:
```
[output_pin LED]
pin: !PB2
value: 1
pwm: false
```
Now when I click on the pin menu I see the following errors:
[image]

### What did you expect to happen instead?

I expected to see a menu with a switch.

### How to reproduce this bug?

I go to the "pin" menu

### Additional information:

I have all the latest updates:
[image]

### Log output

```shell
[main]
language = en
theme = colorized
screen_blanking = off

[graph Printer]
temperature_sensor pi3 = False

=======================
2024-01-16 00:49:38,675 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2024-01-16 00:49:38,838 [config.py:create_translations()] - Selected lang: en OS lang: en_GB
2024-01-16 00:49:38,839 [config.py:install_language()] - Using lang en
2024-01-16 00:49:38,870 [screen.py:__init__()] - Screen resolution: 800x600
2024-01-16 00:49:38,872 [KlippyGtk.py:__init__()] - Font size: 18.6 (medium)
2024-01-16 00:49:39,381 [screen.py:set_screenblanking_timeout()] - Screen blanking: off
2024-01-16 00:49:39,394 [screen.py:initial_connection()] - Default printer: None
2024-01-16 00:49:39,395 [screen.py:connect_printer()] - Connecting to printer: Printer
2024-01-16 00:49:39,395 [screen.py:_load_panel()] - Loading panel: splash_screen
2024-01-16 00:49:39,472 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2024-01-16 00:49:39,486 [KlippyWebsocket.py:connect()] - Attempting to connect
2024-01-16 00:49:39,516 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2024-01-16 00:49:39,535 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2024-01-16 00:49:39,537 [KlippyWebsocket.py:connect()] - Starting websocket thread
2024-01-16 00:49:39,558 [_logging.py:info()] - Websocket connected
2024-01-16 00:49:39,559 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2024-01-16 00:49:39,626 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2024-01-16 00:49:39,645 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/webcams/list
2024-01-16 00:49:39,648 [printer.py:configure_cameras()] - Cameras: [{'name': 'picam', 'enabled': True, 'icon': 'mdiWebcam', 'aspect_ratio': '4:3', 'target_fps': 15, 'target_fps_idle': 5, 'location': 'printer', 'service': 'mjpegstreamer-adaptive', 'stream_url': '/webcam/?action=snapshot', 'snapshot_url': '/webcam/?action=stream', 'flip_horizontal': False, 'flip_vertical': False, 'rotation': 0, 'source': 'database', 'extra_data': {}, 'uid': '76da84ed-c450-475e-9a96-abaed1aad3d3'}]
2024-01-16 00:49:39,648 [screen.py:init_klipper()] - Moonraker info {'klippy_connected': True, 'klippy_state': 'ready', 'components': ['secrets', 'template', 'klippy_connection', 'jsonrpc', 'internal_transport', 'application', 'websockets', 'dbus_manager', 'database', 'file_manager', 'klippy_apis', 'shell_command', 'machine', 'data_store', 'proc_stats', 'job_state', 'job_queue', 'http_client', 'announcements', 'webcam', 'extensions', 'history', 'authorization', 'update_manager', 'octoprint_compat'], 'failed_components': [], 'registered_directories': ['config', 'logs', 'gcodes', 'config_examples', 'docs'], 'warnings': [], 'websocket_count': 3, 'moonraker_version': 'v0.8.0-262-gc4f1d25', 'missing_klippy_requirements': [], 'api_version': [1, 4, 0], 'api_version_string': '1.4.0'}
2024-01-16 00:49:39,670 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/printer/info
2024-01-16 00:49:39,748 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?configfile
2024-01-16 00:49:39,752 [screen.py:init_klipper()] - {'configfile': {'config': {'board_pins mks_rn_11': {'aliases': '\n\nX_STEP_PIN=PE3, X_DIR_PIN=PE2, X_ENABLE_PIN=PE4, X_STOP_PIN=PA15,\nY_STEP_PIN=PE0, Y_DIR_PIN=PB9, Y_ENABLE_PIN=PE1, Y_STOP_PIN=PA12,\nZ_STEP_PIN=PB5, Z_DIR_PIN=PB4, Z_ENABLE_PIN=PB8, Z_STOP_PIN=PA11,\n\nE0_STEP_PIN=PD6, E0_DIR_PIN=PD3, E0_ENABLE_PIN=PB3,\n\nE_HEATER_PIN=PC3, E_SENSOR_PIN=PC1,\nBED_HEATER_PIN=PA0, BED_SENSOR_PIN=PC0,\n\nHEATER_FAN_PIN=PB0,\nPARTCOOLING_FAN_PIN=PB1,\n\nMT_DET1_PIN=PA4, MT_DET2_PIN=PE6,\nPW_DET_PIN=PA2, AUTO_OFF_PIN=PB2,'}, 'temperature_sensor board': {'sensor_type': 'temperature_mcu', 'min_temp': '0', 'max_temp': '100'}, 'output_pin LED': {'pin': '!PB2', 'value': '1', 'pwm': 'false'}, 'mcu': {'serial': '/dev/ttyAMA0', 'restart_method': 'command'}, 'fan': {'pin': 'PB1', 'kick_start_time': '0.200'}, 'heater_fan heater_fan': {'pin': 'PB0'}, 'virtual_sdcard': {'path': '/home/pi/printer_data/gcodes'}, 'filament_switch_sensor filament_sensor': {'switch_pin': 'PA4', 'pause_on_runout': 'True', 'runout_gcode': '\nPAUSE'}, 'output_pin _BEEPER_pin': {'pin': 'PC5', 'pwm': 'True', 'value': '0', 'shutdown_value': '0', 'cycle_time': '0.001'}, 'temperature_sensor Pi3': {'sensor_type': 'temperature_host'}, 'stepper_x': {'step_pin': 'PE3', 'dir_pin': 'PE2', 'enable_pin': '!PE4', 'microsteps': '32', 'rotation_distance': '39.86433', 'endstop_pin': '!PA15', 'position_endstop': '0', 'position_max': '245', 'homing_speed': '40'}, 'stepper_y': {'step_pin': 'PE0', 'dir_pin': 'PB9', 'enable_pin': '!PE1', 'microsteps': '32', 'rotation_distance': '40.035469', 'endstop_pin': '!PA12', 'position_endstop': '-4.3', 'position_max': '207', 'homing_speed': '40', 'position_min': '-4.3'}, 'stepper_z': {'step_pin': 'PB5', 'dir_pin': '!PB4', 'enable_pin': '!PB8', 'position_endstop': '0.09', 'microsteps': '32', 'rotation_distance': '8', 'endstop_pin': '!PA11', 'position_max': '200', 'homing_speed': '5'}, 'extruder': {'step_pin': 'PD6', 'dir_pin': '!PD3', 'enable_pin': '!PB3', 'microsteps': '32', 'rotation_distance': '7.856889085714286', 'nozzle_diameter': '0.4', 'filament_diameter': '1.75', 'max_extrude_cross_section': '0.64', 'max_extrude_only_distance': '150', 'heater_pin': 'PC3', 'sensor_type': 'ATC Semitec 104NT-4-R025H42G', 'sensor_pin': 'PC1', 'control': 'pid', 'pid_kp': '25.674', 'pid_ki': '1.646', 'pid_kd': '100.128', 'min_temp': '0', 'max_temp': '285', 'pressure_advance': '0.159'}, 'heater_bed': {'heater_pin': 'PA0', 'sensor_type': 'EPCOS 100K B57560G104F', 'sensor_pin': 'PC0', 'control': 'pid', 'pid_kp': '58.399', 'pid_ki': '1.570', 'pid_kd': '543.112', 'min_temp': '0', 'max_temp': '110'}, 'printer': {'kinematics': 'cartesian', 'max_velocity': '150', 'max_accel': '3000', 'max_accel_to_decel': '1500', 'max_z_velocity': '20', 'max_z_accel': '100'}, 'bed_screws': {'screw1': '0,18', 'screw1_name': 'front left screw', 'screw2': '205,18', 'screw2_name': 'front right screw', 'screw3': '205,200', 'screw3_name': 'back right screw', 'screw4': '0,200', 'screw4_name': 'back left screw', 'speed': '190', 'horizontal_move_z': '15', 'probe_height': '2'}, 'input_shaper': {'shaper_freq_x': '40.65', 'shaper_freq_y': '37.6', 'shaper_type': 'mzv'}, 'pause_resume': {'recover_velocity': '80'}, 'display_status': {}, 'firmware_retraction': {}, 'respond': {}, 'gcode_macro START_PRINT': {'description': 'Start routine for the print', 'variable_retract': '9', 'gcode': '\nSET_PIN PIN=LED VALUE=1\n{% set extruder_temp = params.EXTRUDER_TEMP|default(230)|float %}\n{% set bed_temp = params.BED_TEMP|default(90)|float %}\n{% set E = printer["gcode_macro START_PRINT"].retract|float %}\nG21\nG90\nM82\nM107\nM27\nCLEAR_PAUSE\nM220 S100\nM221 S100\nG90\nM82\nSET_HEATER_TEMPERATURE HEATER=heater_bed TARGET={bed_temp}\nTEMPERATURE_WAIT SENSOR=heater_bed MINIMUM={bed_temp * 0.6}\nTEMPERATURE_WAIT SENSOR=heater_bed MINIMUM={bed_temp}\nSET_HEATER_TEMPERATURE HEATER=extruder TARGET={extruder_temp}\nTEMPERATURE_WAIT SENSOR=extruder MINIMUM={extruder_temp}\nG28 X0 Y0\nG28 Z0\nG1 Z15.0 F1200\nG92 E0\nG1 F200 E12\nG92 E0\nG1 F7200\nG1 X4.4 Y20 Z0.28 F5000.0\nG1 X4.4 Y170.0 Z0.28 F1500.0 E15\nG92 E0\nM117 Printing'}, 'gcode_macro END_PRINT': {'gcode': '\n{% set E = printer["gcode_macro START_PRINT"].retract|float %}\nTURN_OFF_HEATERS\nM107\nG91\nG1 E-{E} F1500\nG90\nPARK\nM84\nSET_PIN PIN=LED VALUE=0'}, 'gcode_macro PARK': {'gcode': '\n{% set x_park = params.X|default(0)|float %}\n{% set y_park = params.Y|default(0)|float %}\n{% set z_park = params.Z|default(10)|float + printer.toolhead. [... cut ...]
2024-01-16 01:01:07,058 [functions.py:log_start()] - --------------------KlipperScreen Log Start--------------------
2024-01-16 01:01:07,059 [functions.py:log_start()] - KlipperScreen Version: v0.3.8-58-g58d1c04
2024-01-16 01:01:07,059 [functions.py:log_start()] - Python version: 3.9
2024-01-16 01:01:07,250 [config.py:get_config_file_location()] - Passed config (-c): /home/pi/KlipperScreen.conf
2024-01-16 01:01:07,251 [config.py:__init__()] - Config path location: /home/pi/printer_data/config/KlipperScreen.conf
2024-01-16 01:01:07,273 [config.py:log_config()] -  ===== Config File =====
```

closed 2024-01-16 by alfrix

gumileeev, 2024-01-16:
Now I see the slider again, from 0 to 100. I will again get an error when setting intermediate values. For me, only 0 or 100 is true. Shouldn't there be a switch?
```
[output_pin LED]
pin: !PB2
value: 1
pwm: false
```
[image]
[image]

gumileeev, 2024-01-16:
> Attach the full logfile
@alfrix 
[KlipperScreen.log](https://github.com/KlipperScreen/KlipperScreen/files/13947391/KlipperScreen.log)

gumileeev, 2024-01-18:
@alfrix tell me, is there a chance that a switch will appear in the interface instead of a slider?

alfrix, 2024-01-18:
> @alfrix tell me, is there a chance that a switch will appear in the interface instead of a slider?

It should be working, maybe you didn't update to the last commit?

gumileeev, 2024-01-18:
> > @alfrix tell me, is there a chance that a switch will appear in the interface instead of a slider?
> 
> It should be working, maybe you didn't update to the last commit?

Thank you! Now it actually works as expected!

commit 294c1967 (2024-01-16): pins: fix #1242