KlipperScreen/KlipperScreen #1256: [BUG] KlipperScreen.log not created

john-clark, 2024-01-28:
### What happened?

I have used kiauh to install KlipperScreen and ./KlipperScreen/scripts/KlipperScreen-install.sh

find / 

### What did you expect to happen instead?

To find a log someplace.

### How to reproduce this bug?

Look in mainsail logs and it is missing.

### Additional information:

sudo find / -iname "klipperscreen.log"

No results

Nothing in printer_data/logs

### Log output

```shell
pi@moonraker:~ $ ./KlipperScreen/scripts/KlipperScreen-install.sh
Update package data
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
Fetched 15.0 kB in 1s (12.0 kB/s)
Reading package lists... Done
Checking for broken packages...
No broken packages
Installing KlipperScreen dependencies
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
x11-xserver-utils is already the newest version (7.7+8).
xinit is already the newest version (1.4.0-1).
xinput is already the newest version (1.6.3-1).
xserver-xorg-input-evdev is already the newest version (1:2.10.6-2).
xserver-xorg-input-libinput is already the newest version (0.30.0-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Installed X
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
fonts-ipafont is already the newest version (00303-21).
fonts-nanum is already the newest version (20200506-1).
libmpv-dev is already the newest version (0.32.0-3).
network-manager is already the newest version (1.30.6-1+deb11u1).
policykit-1 is already the newest version (0.105-31+rpt1+deb11u1).
xserver-xorg-legacy is already the newest version (2:1.20.11-1+rpt3+deb11u11).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
network-manager
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
xserver-xorg-video-fbdev is already the newest version (1:0.5.0-1+b2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Installed FBdev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-distutils is already the newest version (3.9.2-1).
python3-virtualenv is already the newest version (20.4.0+ds-2+deb11u1).
virtualenv is already the newest version (20.4.0+ds-2+deb11u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Installed Python dependencies
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gcc is already the newest version (4:10.2.1-1+rpi1).
libgirepository1.0-dev is already the newest version (1.66.1-1+b1).
pkg-config is already the newest version (0.29.2-1).
python3-dev is already the newest version (3.9.2-3).
gir1.2-gtk-3.0 is already the newest version (3.24.24-4+rpt12+deb11u3).
libcairo2-dev is already the newest version (1.16.0-5+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Installed PyGobject dependencies
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
autoconf is already the newest version (2.69-14).
libdbus-glib-1-dev is already the newest version (0.110-6).
libopenjp2-7 is already the newest version (2.4.0-3).
librsvg2-common is already the newest version (2.50.3+dfsg-1+deb11u1).
wireless-tools is already the newest version (30~pre9-13.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Installed Misc packages
Checking Python version
Python 3.9.2
Creating virtual environment
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Ignoring python-mpv: markers 'python_version >= "3.10"' don't match your environment
Ignoring backports.zoneinfo: markers 'python_version < "3.9"' don't match your environment
Ignoring PyGObject: markers 'python_version == "3.7"' don't match your environment
Ignoring pycairo: markers 'python_version == "3.7"' don't match your environment
Ignoring websocket-client: markers 'python_version == "3.7"' don't match your environment
Requirement already satisfied: jinja2==3.1.3 in ./.KlipperScreen-env/lib/python3.9/site-packages (from -r /home/pi/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 1)) (3.1.3)
Requirement already satisfied: requests==2.31.0 in ./.KlipperScreen-env/lib/python3.9/site-packages (from -r /home/pi/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 2)) (2.31.0)
Requirement already satisfied: netifaces==0.11.0 in ./.KlipperScreen-env/lib/python3.9/site-packages (from -r /home/pi/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 3)) (0.11.0)
Requirement already satisfied: six==1.16.0 in ./.KlipperScreen-env/lib/python3.9/site-packages (from -r /home/pi/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 4)) (1.16.0)
Requirement already satisfied: dbus-python==1.3.2 in ./.KlipperScreen-env/lib/python3.9/site-packages (from -r /home/pi/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 5)) (1.3.2)
Requirement already satisfied: python-mpv==0.5.2 in ./.KlipperScreen-env/lib/python3.9/site-packages (from -r /home/pi/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 8)) (0.5.2)
Requirement already satisfied: PyGObject==3.46.0 in ./.KlipperScreen-env/lib/python3.9/site-packages (from -r /home/pi/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 18)) (3.46.0)
Requirement already satisfied: pycairo==1.25.1 in ./.KlipperScreen-env/lib/python3.9/site-packages (from -r /home/pi/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 19)) (1.25.1)
Requirement already satisfied: websocket-client==1.7.0 in ./.KlipperScreen-env/lib/python3.9/site-packages (from -r /home/pi/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 20)) (1.7.0)
Requirement already satisfied: MarkupSafe>=2.0 in ./.KlipperScreen-env/lib/python3.9/site-packages (from jinja2==3.1.3->-r /home/pi/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 1)) (2.1.4)
Requirement already satisfied: charset-normalizer<4,>=2 in ./.KlipperScreen-env/lib/python3.9/site-packages (from requests==2.31.0->-r /home/pi/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 2)) (3.3.2)
Requirement already satisfied: urllib3<3,>=1.21.1 in ./.KlipperScreen-env/lib/python3.9/site-packages (from requests==2.31.0->-r /home/pi/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 2)) (2.1.0)
Requirement already satisfied: certifi>=2017.4.17 in ./.KlipperScreen-env/lib/python3.9/site-packages (from requests==2.31.0->-r /home/pi/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 2)) (2023.11.17)
Requirement already satisfied: idna<4,>=2.5 in ./.KlipperScreen-env/lib/python3.9/site-packages (from requests==2.31.0->-r /home/pi/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 2)) (3.6)
Virtual enviroment created
Installing KlipperScreen PolicyKit Rules
PolicyKit Version 0.105 Detected
Updating X11 Xwrapper
Install as a service? (This will enable boot to console) [Y/n]y
Installing KlipperScreen unit file
Starting service...
```

alfrix, 2024-01-29:
this is not a bug if klipperscreen could not start you will not find the log, it's described on the docs
https://klipperscreen.github.io/KlipperScreen/Troubleshooting/#system-logs

closed 2024-01-29 by alfrix

john-clark, 2024-01-29:
I thought that the KlipperScreen install scrip[t should have left a log.

The solution was `usermod -a -G tty ${username}` and  `sudo bash -c "echo needs_root_rights=yes>>/etc/X11/Xwrapper.config"` as noted in the [documentation](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/VC_ERROR/).

I haven't taken a look at the install script but issue this could be corrected programmatically, and should generate a code correction. I will take a look when I have a chance.