KlipperScreen/KlipperScreen #1261: [FR] Support POWER_OFF_PRINTER

foosinn, 2024-02-04:
### Requested feature

Moonraker has a function to power off the printer when idle.

It would be awesome of KlipperScreen would detect that the printer turned off (see additonal information).

### Possible workflows:

**power off event received**:

Turn off display and switch to a new panel that provides nothing but a power on button.

**power on event received**

Turn on screen and switch to default panel

**power on button is pressed**

Send power on command (see additional information) and switch to default panel

### Is your feature request related to a problem?

Currently KlipperScreen is just displaying that klipper is disconnected when the printer is powered down

### Describe alternatives you have considered

There are no other ways to display a power on button on the screen

### Additional information

Moonraker Docs:
https://moonraker.readthedocs.io/en/latest/configuration/#toggling-device-state-from-klipper

Klipper Screen is already receiving the relevent events. 

power on status:
{"jsonrpc": "2.0", "method": "notify_power_changed", "params": [{"device": "printer", "status": "on", "locked_while_printing": true, "type": "homeassistant"}]}

power off status:
{"jsonrpc": "2.0", "method": "notify_power_changed", "params": [{"device": "printer", "status": "off", "locked_while_printing": true, "type": "homeassistant"}]}

power on command:
{"jsonrpc":"2.0","method":"machine.device_power.post_device","params":{"device":"printer","action":"on"},"id":285}

alfrix, 2024-02-04:
you need to define the moonraker power device in klipperscreen printer options, to have the turn on printer in splash-screen, also the power panel can turn the printer on

https://klipperscreen.github.io/KlipperScreen/Configuration/#printer-options

close as already implemented

closed 2024-02-04 by alfrix

foosinn, 2024-02-06:
Hey, i've experimented quite a bit with it, but it seems like I cant get it to work.

I have setup the `power_devices` section like you mentioned. Still KlipperScreen stays active when I run `POWER_OFF_PRINTER`.

Any hint what could be missing? Thanks!

alfrix, 2024-02-12:
> Hey, i've experimented quite a bit with it, but it seems like I cant get it to work.
> 
> I have setup the `power_devices` section like you mentioned. Still KlipperScreen stays active when I run `POWER_OFF_PRINTER`.
> 
> Any hint what could be missing? Thanks!

the power_devices setting function is creating a button in the splash screen

if you want to turn off the screen with a macro then you would need to use gcode_shell_command extension (use kiauh)

for example to turn the screen off you could add the command `DISPLAY:=0 xset dpms force off`
the same if you want to power off the pi the command would be `systemctl poweroff` 
this kind of functionality will not be part of klipperscreen since it should be done with macros

foosinn, 2024-02-12:
Thanks a lot for the update. 

`gcode_shell_command` is not an option for me, since the printer is in a shared space, i was able to solve it.

I would love to have a config option for that, but if you do not plan to support i also can keep using this little patch.

```diff
diff --git a/screen.py b/screen.py
index aebbb48..6b8cb9b 100755
--- a/screen.py
+++ b/screen.py
@@ -344,7 +344,7 @@ class KlipperScreen(Gtk.Window):
         if (datetime.now() - self.last_popup_time).seconds < 1:
             return
         self.last_popup_time = datetime.now()
-        self.close_screensaver()
+        # should be only required while KlipperScreen is using unsupported Moonraker options
+        #self.close_screensaver()
         if self.popup_message is not None:
             self.close_popup_message()

@@ -777,6 +777,10 @@ class KlipperScreen(Gtk.Window):
             logging.debug("Power status changed: %s", data)
             self.printer.process_power_update(data)
             self.panels['splash_screen'].check_power_status()
+            if data['status'] == "off":
+                os.system("xset -display :0 dpms force off")
+            else:
+                os.system("xset -display :0 dpms force on")
         elif action == "notify_gcode_response" and self.printer.state not in ["error", "shutdown"]:
             if not (data.startswith("B:") or data.startswith("T:")):
                 if data.startswith("// action:"):
```