KlipperScreen/KlipperScreen #1273: [BUG]  console and notification pages are not  rendered correctly on ubuntu server 23.10 with raspberry pi 4

ss8688, 2024-02-15:
### What happened?

console and notification pages seem not  rendered correctly.They look very strange.
[image]

[image]

[image]

### What did you expect to happen instead?

words can see clearly

### How to reproduce this bug?

runing klipperscreen on on ubuntu server 23.10 with raspberry pi 4

### Log output

```shell
body is too long (maximum is 65536 characters). So offer logs latter
```

ss8688, 2024-02-15:
```
[main]
language = zh_CN

=======================
2024-02-15 10:54:37,119 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2024-02-15 10:54:37,189 [config.py:create_translations()] - Selected lang: zh_CN OS lang: zh_CN
2024-02-15 10:54:37,190 [config.py:install_language()] - Using lang zh_CN
2024-02-15 10:54:37,212 [screen.py:__init__()] - Screen resolution: 800x480
2024-02-15 10:54:37,213 [KlippyGtk.py:__init__()] - Font size: 16.6 (medium)
2024-02-15 10:54:37,578 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 3600
2024-02-15 10:54:37,592 [screen.py:set_screenblanking_timeout()] - Using DPMS
2024-02-15 10:54:37,610 [screen.py:initial_connection()] - Default printer: None
2024-02-15 10:54:37,650 [screen.py:connect_printer()] - Connecting to printer: Printer
2024-02-15 10:54:37,651 [screen.py:_load_panel()] - Loading panel: splash_screen
2024-02-15 10:54:37,704 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2024-02-15 10:54:37,713 [KlippyWebsocket.py:connect()] - Attempting to connect
2024-02-15 10:54:37,725 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2024-02-15 10:54:37,734 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2024-02-15 10:54:37,736 [KlippyWebsocket.py:connect()] - Starting websocket thread
2024-02-15 10:54:37,744 [_logging.py:info()] - Websocket connected
2024-02-15 10:54:37,745 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2024-02-15 10:54:37,874 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/info
2024-02-15 10:54:37,882 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/webcams/list
2024-02-15 10:54:37,882 [printer.py:configure_cameras()] - Cameras: [{'name': 'camera0', 'enabled': True, 'icon': 'mdiWebcam', 'aspect_ratio': '16:9', 'target_fps': 30, 'target_fps_idle': 30, 'location': 'printer', 'service': 'mjpegstreamer-adaptive', 'stream_url': '/webcam/?action=stream', 'snapshot_url': '/webcam/?action=snapshot', 'flip_horizontal': False, 'flip_vertical': False, 'rotation': 0, 'source': 'config', 'extra_data': {}, 'uid': 'c3746231-7e67-5c0e-8f05-1e7bab1f89d2'}]
2024-02-15 10:54:37,883 [screen.py:init_klipper()] - Moonraker info {'klippy_connected': True, 'klippy_state': 'ready', 'components': ['secrets', 'template', 'klippy_connection', 'jsonrpc', 'internal_transport', 'application', 'websockets', 'dbus_manager', 'database', 'file_manager', 'authorization', 'klippy_apis', 'shell_command', 'machine', 'data_store', 'proc_stats', 'job_state', 'job_queue', 'http_client', 'announcements', 'webcam', 'extensions', 'octoprint_compat', 'history', 'update_manager'], 'failed_components': [], 'registered_directories': ['config', 'logs', 'gcodes', 'config_examples', 'docs'], 'warnings': [], 'websocket_count': 8, 'moonraker_version': 'v0.8.0-310-g67c98f6', 'missing_klippy_requirements': [], 'api_version': [1, 4, 0], 'api_version_string': '1.4.0'}
2024-02-15 10:54:37,895 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/printer/info
2024-02-15 10:54:38,087 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?configfile
2024-02-15 10:54:38,089 [screen.py:init_klipper()] - {'configfile': {'config': {'virtual_sdcard': {'path': '~/printer_data/gcodes', 'on_error_gcode': 'CANCEL_PRINT'}, 'pause_resume': {}, 'display_status': {}, 'respond': {}, 'gcode_macro CANCEL_PRINT': {'description': 'Cancel the actual running print', 'rename_existing': 'CANCEL_PRINT_BASE', 'gcode': '\n\n{% set client = printer[\'gcode_macro _CLIENT_VARIABLE\']|default({}) %}\n{% set allow_park = client.park_at_cancel|default(false)|lower == \'true\' %}\n{% set retract = client.cancel_retract|default(5.0)|abs %}\n\n{% set park_x = "" if (client.park_at_cancel_x|default(none) is none)\nelse "X=" ~ client.park_at_cancel_x %}\n{% set park_y = "" if (client.park_at_cancel_y|default(none) is none)\nelse "Y=" ~ client.park_at_cancel_y %}\n{% set custom_park = park_x|length > 0 or park_y|length > 0 %}\n\n\n{% if printer[\'gcode_macro RESUME\'].restore_idle_timeout > 0 %}\nSET_IDLE_TIMEOUT TIMEOUT={printer[\'gcode_macro RESUME\'].restore_idle_timeout}\n{% endif %}\n{% if (custom_park or not printer.pause_resume.is_paused) and allow_park %} _TOOLHEAD_PARK_PAUSE_CANCEL {park_x} {park_y} {% endif %}\n_CLIENT_RETRACT LENGTH={retract}\nTURN_OFF_HEATERS\nM106 S0\n{client.user_cancel_macro|default("")}\nSET_GCODE_VARIABLE MACRO=RESUME VARIABLE=idle_state VALUE=False\n\nSET_PAUSE_NEXT_LAYER ENABLE=0\nSET_PAUSE_AT_LAYER ENABLE=0 LAYER=0\nCANCEL_PRINT_BASE'}, 'gcode_macro PAUSE': {'description': 'Pause the actual running print', 'rename_existing': 'PAUSE_BASE', 'gcode': '\n\n{% set client = printer[\'gcode_macro _CLIENT_VARIABLE\']|default({}) %}\n{% set idle_timeout = client.idle_timeout|default(0) %}\n{% set temp = printer[printer.toolhead.extruder].target if printer.toolhead.extruder != \'\' else 0 %}\n{% set restore = False if printer.toolhead.extruder == \'\'\nelse True  if params.RESTORE|default(1)|int == 1 else False %}\n\nSET_GCODE_VARIABLE MACRO=RESUME VARIABLE=last_extruder_temp VALUE="{{\'restore\': restore, \'temp\': temp}}"\n\n{% if idle_timeout > 0 %}\nSET_GCODE_VARIABLE MACRO=RESUME VARIABLE=restore_idle_timeout VALUE={printer.configfile. [... cut ...]
2024-02-15 10:54:44,063 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2024-02-15 10:54:44,078 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/temperature_store
2024-02-15 10:54:44,082 [printer.py:init_temp_store()] - Temp store: ['heater_bed', 'extruder']
2024-02-15 10:54:44,092 [KlippyRest.py:_do_request()] - Sending request to http://127.0.0.1:7125/server/config
2024-02-15 10:54:44,092 [screen.py:init_tempstore()] - Temperature store size: 1200
```

alfrix, 2024-02-15:
duplicate of #559 supposedly not happening anymore, i don't think this as bug of klipperscreen, it seems to be a driver issue, it should be [reported to mesa](https://docs.mesa3d.org/bugs.html)

closed 2024-07-14 by alfrix