KlipperScreen/KlipperScreen #1275: [BUG] Installing KlipperScreen inside a chroot (CustomPIOS) fails

KwadFan, 2024-02-17:
### What happened?

During developing of a KlipperScreen Module for our Buildchain, I am not able to install KlipperScreen on Armbian based Images (they use py3.11.2). Everytime I try, it fails at the point building Ninja.

I already tried to install `ninja-build` package but this doesnt help at all.

Here is the full Output of the workflow run: https://github.com/KwadFan/MainsailOS-dev/actions/runs/7940003744/job/21680758799#step:3:21673

Building an Image using a Raspaberry Pi OS is possible and runs just fine.
As you can see here: https://github.com/mainsail-crew/MainsailOS/actions/runs/7855908040/job/21438108889#step:3:10861

So, a hint what could be worng here, would be great.
Regards Kwad
 

### What did you expect to happen instead?

Building an image ...

### How to reproduce this bug?

Fork mainsail-crew/mainsailOS, run a build workflow with enabled KlipperScreen Module.

### Log output

```shell
[16/19] Running Ninja test suite
            FAILED: build_ninja-prefix/src/build_ninja-stamp/build_ninja-run_ninja_test_suite /tmp/pip-install-t3y_ngos/ninja_611ff158dc9d4a2bb48b820cd00cfeb3/_skbuild/linux-armv7l-3.11/cmake-build/build_ninja-prefix/src/build_ninja-stamp/build_ninja-run_ninja_test_suite
            cd /tmp/pip-install-t3y_ngos/ninja_611ff158dc9d4a2bb48b820cd00cfeb3/_skbuild/linux-armv7l-3.11/cmake-build/Ninja-build && /tmp/pip-install-t3y_ngos/ninja_611ff158dc9d4a2bb48b820cd00cfeb3/_skbuild/linux-armv7l-3.11/cmake-build/Ninja-build/ninja_test && /usr/bin/cmake -E touch /tmp/pip-install-t3y_ngos/ninja_611ff158dc9d4a2bb48b820cd00cfeb3/_skbuild/linux-armv7l-3.11/cmake-build/build_ninja-prefix/src/build_ninja-stamp/build_ninja-run_ninja_test_suite
            ninja: build stopped: subcommand failed.
            Traceback (most recent call last):
              File "/tmp/pip-build-env-3spi9ese/overlay/lib/python3.11/site-packages/skbuild/setuptools_wrap.py", line 674, in setup
                cmkr.make(make_args, install_target=cmake_install_target, env=env)
              File "/tmp/pip-build-env-3spi9ese/overlay/lib/python3.11/site-packages/skbuild/cmaker.py", line 697, in make
                self.make_impl(clargs=clargs, config=config, source_dir=source_dir, install_target=install_target, env=env)
              File "/tmp/pip-build-env-3spi9ese/overlay/lib/python3.11/site-packages/skbuild/cmaker.py", line 742, in make_impl
                raise SKBuildError(msg)
      
            An error occurred while building with CMake.
              Command:
                /usr/bin/cmake --build . --target install --config Release --
              Install target:
                install
              Source directory:
                /tmp/pip-install-t3y_ngos/ninja_611ff158dc9d4a2bb48b820cd00cfeb3
              Working directory:
                /tmp/pip-install-t3y_ngos/ninja_611ff158dc9d4a2bb48b820cd00cfeb3/_skbuild/linux-armv7l-3.11/cmake-build
            Please check the install target is valid and see CMake's output for more information.
      
            [end of output]
      
        note: This error originates from a subprocess, and is likely not a problem with pip.
        ERROR: Failed building wheel for ninja
      Failed to build ninja
      ERROR: Could not build wheels for ninja, which is required to install pyproject.toml-based projects
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
Unable to install dependencies, aborting install.
```

alfrix, 2024-02-17:
try installing an older version of ninja

i'll see what i can find after work (yes on saturday and sometimes i have to work sundays :-1:)

KwadFan, 2024-02-17:
Take your time, we all have to work and most 'klipperversum' components are spare time projects

alfrix, 2024-02-17:
lets see:
```
Collecting dbus-python==1.3.2
  Downloading dbus-python-1.3.2.tar.gz (605 kB)
  ...
  Installing build dependencies: finished with status 'error'
  error: subprocess-exited-with-error
  ....
   Error: pip install exited with status code 0
  Trying again with new tools...
 × Building wheel for ninja (pyproject.toml) did not run successfully.
  ...
```
from some searches i can see that some say the issue started appearing with ninja 1.10.2.4
so trying with ninja 1.10.2.3 or older may work

KwadFan, 2024-02-18:
Oh great, thanks, will try this branch today!

KwadFan, 2024-02-18:
So I tried your fix branch, but unfortunatly it didnt work.
If you want to have a look at, it fails after `Installed misc packages`
What drives me nuts that it builds fine on a running system :man_shrugging: 

Here ist the log: https://github.com/KwadFan/MainsailOS-dev/actions/runs/7948120276/job/21697775111#step:3:20951

alfrix, 2024-02-18:
it does install ninja into the env successfully, so this older version works, but it doesn't want to use it to build the dbus package, i assume that is looking for that package in a global context (`ninja-build`), but you said it didn't work?

i checked the dependency ant it doesn't require a specific version to build

```
Successfully installed ninja-1.10.0
...
Collecting ninja
Downloading ninja-1.11.1.1.tar.gz (132 kB)
Installing build dependencies: started
...
Building wheels for collected packages: ninja
 × Building wheel for ninja (pyproject.toml) did not run successfully.
```

what bothers me is that this specific package (dbus-python) is part of the old backend that is already marked for deprecation  #1269

alfrix, 2024-02-18:
there is an alternative to install a package globally instead of ninja-build 
`pip install ninja=1.10.0 --break-system-packages`
that needs to run before the ks install

KwadFan, 2024-02-20:
I am sorry, got stuck in rewriten parts of buildchain....

So, I will try your latest recommendation, but I have some concerns that it will break if system package recieves an update.
Also, wasn't there a change that pip is not allowed to globally install packages?
(But I am no py dev so I might be wrong here)
You mentioned #1269 wouldnt it make sense to make a new branch with that PR and you changes for this issue just for testing?
Regards Kwad

alfrix, 2024-02-20:
> I am sorry, got stuck in rewriten parts of buildchain....
>
> So, I will try your latest recommendation, but I have some concerns that it will break if system package recieves an update. Also, wasn't [...]

pip should be used globally, the command should not be used on a production image, it was meant to see if the old version fixes it and work from there to a solution (create a wheel for the package > save this wheel somewhere > set the buildchain to download and install the prebuilt wheel directly)

>You mentioned #1269 wouldnt it make sense to make a new branch with that PR and you changes for this issue just for testing? Regards Kwad

if you want to test that, the repo is https://github.com/alfrix/KlipperScreen
and the branch is new_network_backend
i will update it to the last version i have, but that is scheduled to be merged with the drop of python 3.7

KwadFan, 2024-03-03:
Well, non of these workarounds worked, but after packing my backpack, grabbing my shovel and digging deeper into that rabbithole, I discovered the same issue for Klippers Input Shaper requirements, formerly numpy.

Thinking about and reading the error output dozen times, it seems some distributions are not precofigured to use piwheels.org. That said, would you mind, if I open a PR to add `--extra-index-url` to flags if not listed in `/etc/pip.conf` ?

Regards Kwad

KwadFan, 2024-03-03:
Here is my Proof-of-Concept:

https://github.com/KwadFan/KlipperScreen/tree/fix/chrootinstall

https://github.com/KwadFan/KlipperScreen/blob/fix/chrootinstall/scripts/KlipperScreen-install.sh#L103-L138

Would you allow this change?
In chroot build it works fine, need to check "live" install but should overall speed up install process on 32bit ARM Sbc due usage of prebuild pip packages from piwheels.

Regards Kwad

KwadFan, 2024-03-03:
Adding some Log output fyi:
```Installed Misc packages
Checking Python version
Python 3.11.2
Creating virtual environment
created virtual environment CPython3.11.2.final.0-32 in 740ms
  creator CPython3Posix(dest=/home/pi/.KlipperScreen-env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/pi/.local/share/virtualenv)
    added seed packages: pip==23.0.1, setuptools==66.1.1, wheel==0.38.4
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Using armv[67]l! Adding piwheels.org as extra index...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Ignoring python-mpv: markers 'python_version < "3.10"' don't match your environment
Ignoring backports.zoneinfo: markers 'python_version < "3.9"' don't match your environment
Ignoring PyGObject: markers 'python_version == "3.7"' don't match your environment
Ignoring pycairo: markers 'python_version == "3.7"' don't match your environment
Ignoring websocket-client: markers 'python_version == "3.7"' don't match your environment
Collecting jinja2==3.1.3
  Downloading https://www.piwheels.org/simple/jinja2/Jinja2-3.1.3-py3-none-any.whl (133 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 133.2/133.2 kB 768.1 kB/s eta 0:00:00
Collecting requests==2.31.0
  Downloading https://www.piwheels.org/simple/requests/requests-2.31.0-py3-none-any.whl (62 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.6/62.6 kB 707.7 kB/s eta 0:00:00
Collecting netifaces==0.11.0
  Downloading https://www.piwheels.org/simple/netifaces/netifaces-0.11.0-cp311-cp311-linux_armv7l.whl (33 kB)
Collecting six==1.16.0
  Downloading https://www.piwheels.org/simple/six/six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting dbus-python==1.3.2
  Downloading https://www.piwheels.org/simple/dbus-python/dbus_python-1.3.2-cp311-cp311-linux_armv7l.whl (355 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 355.3/355.3 kB 2.2 MB/s eta 0:00:00
Collecting python-mpv==1.0.5
  Downloading https://www.piwheels.org/simple/python-mpv/python_mpv-1.0.5-py3-none-any.whl (44 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 44.6/44.6 kB 5.6 MB/s eta 0:00:00
Collecting PyGObject==3.46.0
  Downloading https://www.piwheels.org/simple/pygobject/PyGObject-3.46.0-cp311-cp311-linux_armv7l.whl (766 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 766.6/766.6 kB 7.5 MB/s eta 0:00:00
Collecting pycairo==1.26.0
  Downloading https://www.piwheels.org/simple/pycairo/pycairo-1.26.0-cp311-cp311-linux_armv7l.whl (317 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 317.4/317.4 kB 10.2 MB/s eta 0:00:00
Collecting websocket-client==1.7.0
  Downloading https://www.piwheels.org/simple/websocket-client/websocket_client-1.7.0-py3-none-any.whl (58 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 58.5/58.5 kB 7.1 MB/s eta 0:00:00
Collecting MarkupSafe>=2.0
  Downloading https://www.piwheels.org/simple/markupsafe/MarkupSafe-2.1.5-cp311-cp311-linux_armv7l.whl (26 kB)
Collecting charset-normalizer<4,>=2
  Downloading https://www.piwheels.org/simple/charset-normalizer/charset_normalizer-3.3.2-py3-none-any.whl (48 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 48.5/48.5 kB 486.8 kB/s eta 0:00:00
Collecting idna<4,>=2.5
  Downloading https://www.piwheels.org/simple/idna/idna-3.6-py3-none-any.whl (61 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.6/61.6 kB 699.8 kB/s eta 0:00:00
Collecting urllib3<3,>=1.21.1
  Downloading https://www.piwheels.org/simple/urllib3/urllib3-2.2.1-py3-none-any.whl (121 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 121.1/121.1 kB 736.7 kB/s eta 0:00:00
Collecting certifi>=2017.4.17
  Downloading https://www.piwheels.org/simple/certifi/certifi-2024.2.2-py3-none-any.whl (163 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 163.8/163.8 kB 2.1 MB/s eta 0:00:00
Installing collected packages: netifaces, websocket-client, urllib3, six, python-mpv, pycairo, MarkupSafe, idna, dbus-python, charset-normalizer, certifi, requests, PyGObject, jinja2
Successfully installed MarkupSafe-2.1.5 PyGObject-3.46.0 certifi-2024.2.2 charset-normalizer-3.3.2 dbus-python-1.3.2 idna-3.6 jinja2-3.1.3 netifaces-0.11.0 pycairo-1.26.0 python-mpv-1.0.5 requests-2.31.0 six-1.16.0 urllib3-2.2.1 websocket-client-1.7.0
Virtual enviroment created```

alfrix, 2024-03-03:
> Would you allow this change?

yes

KwadFan, 2024-03-03:
Cool, will prepare PR for it...

PR #1291 (2024-03-04): fix: fix install in chroot
This fixes #1275 .

This should also speed up install on armv7l/armv6l devices due the fact it uses precompiled packages.
closed 2024-03-04 by alfrix