KlipperScreen/KlipperScreen #1282: [BUG] <title> Folders are not shown in the directory with G-codes.

user32rtu, 2024-02-21:
### What happened?

I turn on the printer and the system boots. I go to the print file selection panel and start printing. After printing is complete, I go back to the print file selection panel, but no folders are displayed. After rebooting the system, the folders appear again.

### What did you expect to happen instead?

Folders should always be visible in the print file selection panel.

### How to reproduce this bug?

1. When the printer is turned on, start printing.
2. After printing is complete, select another file.
3. Folders are not shown in the selection panel.

### Additional information:

Sapphire Pro. 
[image]

### Log output

```shell
[KlipperScreen.log](https://github.com/KlipperScreen/KlipperScreen/files/14362755/KlipperScreen.log)
```

CIS94500, 2024-02-23:
hi user32rtu, 

this modification fixed the problem for me

in file :
```
KlipperScreen/panels/print.py
```

move line 358 to 365
```
self.filelist = {'gcodes': {'directories': [], 'files': []}}
```
before move :

```
    def reload_files(self, widget=None):
        self.filelist = {'gcodes': {'directories': [], 'files': []}}
        for dirpan in self.dir_panels:
            for column in range(3):
                self.dir_panels[dirpan].remove_column(column)
        self.load_files()

    def load_files(self):
        flist = sorted(self._screen.files.get_file_list(), key=lambda item: '/' in item)
        for file in flist:
            self.add_file(file)
        self.show_directory()
```		
		
after move :

```
    def reload_files(self, widget=None):
        for dirpan in self.dir_panels:
            for column in range(3):
                self.dir_panels[dirpan].remove_column(column)
        self.load_files()

    def load_files(self):
	self.filelist = {'gcodes': {'directories': [], 'files': []}}
        flist = sorted(self._screen.files.get_file_list(), key=lambda item: '/' in item)
        for file in flist:
            self.add_file(file)
        self.show_directory()
```

user32rtu, 2024-02-24:
Hello CIS94500. Once the printing is finished, I'll try your method.

user32rtu, 2024-02-24:
> Hello CIS94500. Once the printing is finished, I'll try your method.

Unfortunately, this didn't help.

user32rtu, 2024-02-24:
> Hello CIS94500. Once the printing is finished, I'll try your method.

Yes, it worked. Thank you !

PR #1270 (2024-02-29): New print panel
new features:
* sort by size
* view as thumbnails
* initial load is faster, particularly if the gcode folder has a lot of subfolders
Can be improved:
* support for filtering/search entry is possible now

close #1282
closed 2024-02-29 by alfrix