KlipperScreen/KlipperScreen #1289: [FR] Wayland Standalone (Kiosk)

alfrix, 2024-03-02:
### Requested feature

Initial Testing

install needed packages
```
sudo apt install cage seatd xwayland
```

> [!NOTE]
> Apparently cage needs xwayland, except in the master branch (0.1.6 ?)
> https://github.com/cage-kiosk/cage/commit/34eb3ec2c81fde3349eed63daba8b244b0bfd46f

In scripts/launch_KlipperScreen.sh:
```sh
export XDG_RUNTIME_DIR=/run/user/$(id -u)
cage -d $KS_XCLIENT
```

> [!NOTE]
> $KS_XCLIENT = ~/.KlippeScreen-env/python ~/KlippeScreen/screen.py

reboot

KwadFan, 2024-03-03:
Just my 2 cents:

If cage ( I know its a kind of chroot jail for X application to stick in fullscreen kiosk mode ), uses Xwayland, its not really using wayland only a layer more to have X compatibilty on Wayland... Which in the end adds a layer of complexity to devices that are based on Xorg rather than Wayland (even if there are Wayland packages available)

Worth it?

alfrix, 2024-03-03:
> Just my 2 cents:
>
> If cage ( I know its a kind of chroot jail for X application to stick in fullscreen kiosk mode ), uses Xwayland, its not really using wayland only a layer more to have X [...]

first off, doing the steps i wrote, KlippeScreen will does run on wayland and not xwayland. 
i've tested using a kms/drm driver, (required by wayland according to my research)

it seems that xWayland was required by cage to allow applications that require X (compatibility reasons)
I've tried building  `Cage version 0.1.5-34eb3ec (branch 'master')`
and it doesn't need xwayland to start, (i removed the package from the system to be 100% sure)
but bookworm will likely remain at cage v0.1.4

this is still experimental, here are my notes so far:
gles fails on my hw setup and i have to use pixman as renderer
building cage from the master branch will fail due to dependency of wlroots being too new (requires 0.17)
building cage using the first commit that doesn't require xwayland works, but will image will freeze after a while

> 
> Worth it?

probably not, but x will be deprecated, it's better to prepared, there is no rush at the moment

PR #1306 (2024-04-03): Add an alternative to X close #1289
close #1289
commit 6cd1a8e1 (2024-04-03): Add an alternative to X close #1289 (#1306)

* Changes to allow using cage-kiosk

Wayland support close #1289

* installer: many improvements

* installer: add notes about wayland status

* deps: don't pull xserver with updates
closed 2024-04-03 by alfrix