KlipperScreen/KlipperScreen #1290: Cannot install X-server dependency

pinzarufilip, 2024-03-03:
### What happened?

tried to install Klipperscreen on fluiddpi

### What did you expect to happen instead?

Install Klipperscreen

### How to reproduce this bug?

instal on fluiddpi Klipperscreen.

### Log output

```shell
Selecting previously unselected package libegl-mesa0:armhf.
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'ocl-icd-libopencl1:armhf' contains empty filename
E: Sub-process /usr/bin/dpkg returned an error code (2)
Installation of X-server dependencies failed (xinit xinput x11-xserver-utils xserver-xorg-input-evdev xserver-xorg-input-libinput)

#=======================================================#
 KlipperScreen installation failed!
#=======================================================#
```

alfrix, 2024-03-03:
do not use FluiddPI

[image]

closed 2024-03-03 by alfrix

CybrGost69, 2025-05-06:
same thing JUST happened to me while installing KlipperScreen via KIAUH.