KlipperScreen/KlipperScreen #1294: Does not show print time in file information.

user32rtu, 2024-03-05:
### What happened?

In KlipperScreen v0.3.9-1, printing time is not shown when selecting a file.

### What did you expect to happen instead?

In previous versions, the print time was visible when selecting a file.

### How to reproduce this bug?

Select file to print.

### Log output

```shell
[logs-20240305-200501.zip](https://github.com/KlipperScreen/KlipperScreen/files/14498857/logs-20240305-200501.zip)
```

alfrix, 2024-03-07:
do you mean that you want to change the view mode to list? it's on the top right corner....

user32rtu, 2024-03-10:
> do you mean that you want to change the view mode to list? it's on the top right corner....
No, in list mode there is no information about printing time either. Print times were shown in previous versions of KlipperScreen.

CIS94500, 2024-03-13:
hi user32rtu,

if the problem is that the printing time does not appear when you are in a folder, try this :

in file
```
panels/print.py
```

line 147

replace
```
info.set_markup(self.get_info_str(item))
```
to
```
info.set_markup(self.get_info_str(item, path))
```

line 334

replace
```
def get_info_str(self, item):
```
to
```
def get_info_str(self, item, path):
```

line 345

replace

```
fileinfo = self._screen.files.get_file_info(item['filename'])
```
to
```
fileinfo = self._screen.files.get_file_info(path)
```

commit 46f3f80c (2024-03-15): print: fix print time not showing in subfolders fixes #1294