KlipperScreen/KlipperScreen #1302: [BUG] Wan SSID missing in wifi pannel

umeiko, 2024-03-12:
### What happened?

I can't found my ssid on the KilpperScreen's Wifi pannel, but when i use `nmcli dev wifi` command, I can see my ssid.

### What did you expect to happen instead?

This is `nmcli dev wifi` shows:
    umeko@umeko-hm2:~$ nmcli dev wifi
    IN-USE  BSSID              SSID                          MODE   CHAN  RATE     >
            48:7D:2E:BB:72:99  218-3                         Infra  6     405 Mbit/>
            9A:97:CC:69:39:0A  --                            Infra  153   270 Mbit/>
            98:97:CC:59:39:0A  218-2-5G                      Infra  153   270 Mbit/>
            58:D9:D5:72:69:A8  Tenda_7269A8                  Infra  4     130 Mbit/>
            F4:83:CD:BC:E7:A7  moyu                          Infra  6     405 Mbit/>
            78:44:FD:F1:1D:F6  pelab_peiqi_5G                Infra  149   540 Mbit/>
            18:F2:2C:B8:66:0E  TP-LINK_660E                  Infra  6     270 Mbit/>
            88:C3:97:C0:B5:C2  doczhu-2.4G                   Infra  11    270 Mbit/>
            00:5C:C2:16:60:EA  pelab_201                     Infra  1     270 Mbit/>
            C2:18:03:9A:14:40  DIRECT-40-HP Laser 136w       Infra  1     65 Mbit/s>
            00:5C:C2:16:60:EC  pelab_201                     Infra  153   270 Mbit/>
            94:D9:B3:1A:CF:51  TP-LINK_CF51                  Infra  6     405 Mbit/>
            C4:8E:8F:84:9B:C2  HP-Print-C2-LaserJet Pro MFP  Infra  6     65 Mbit/s>
            98:97:CC:8A:B4:68  Miss.MR                       Infra  48    540 Mbit/>
            9E:97:CC:8A:B4:68  --                            Infra  48    540 Mbit/>
            78:44:FD:F1:1D:F4  pelab_peiqi                   Infra  1     270 Mbit/>
            FA:D0:27:61:04:41  DIRECT-27618441               Infra  6     65 Mbit/s>
            A8:BD:27:02:B8:31  XMUNET+                       Infra  149   270 Mbit/>
            A8:BD:27:02:B8:30  eduroam                       Infra  149   130 Mbit/>
            14:75:90:98:68:A4  309                           Infra  6     405 Mbit/>
            94:D9:B3:1A:CF:53  TP-LINK_5G_CF51               Infra  161   270 Mbit/>
            88:C3:97:C0:B5:C3  doczhu_5G                     Infra  36    540 Mbit/>
But KlipperScreen's pannel shows under, I can't found my ssid `218-3`:
[image]

### How to reproduce this bug?

Maybe set your ssid `218-3` and scan may  reproduce this bug?

### Additional information:

Ubuntu20.04

### Log output

```shell
2023-12-04 19:58:29,893 [functions.py:log_start()] - --------------------KlipperScreen Log Start--------------------
2023-12-04 19:58:29,893 [functions.py:log_start()] - KlipperScreen Version: ?
2023-12-04 19:58:29,893 [functions.py:log_start()] - Python version: 3.8
2023-12-04 19:58:30,012 [config.py:get_config_file_location()] - Passed config (-c): .
2023-12-04 19:58:30,015 [config.py:__init__()] - Config path location: /home/umeko/KlipperScreen/ks_includes/defaults.conf
2023-12-04 19:58:30,026 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2023-12-04 19:58:30,124 [config.py:create_translations()] - Selected lang: None OS lang: zh_CN
2023-12-04 19:58:30,124 [config.py:install_language()] - Using lang zh_CN
2023-12-04 19:58:30,146 [screen.py:__init__()] - Monitors: 1 using number: 0
2023-12-04 19:58:30,149 [screen.py:__init__()] - Screen resolution: 1080x1920
2023-12-04 19:58:30,150 [KlippyGtk.py:__init__()] - Font size: 32.7 (medium)
2023-12-04 19:58:30,589 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 3600
2023-12-04 19:58:30,599 [screen.py:set_screenblanking_timeout()] - Using DPMS
2023-12-04 19:58:30,607 [screen.py:initial_connection()] - Default printer: None
2023-12-04 19:58:30,647 [screen.py:connect_printer()] - Connecting to printer: Printer
2023-12-04 19:58:30,648 [screen.py:_load_panel()] - Loading panel: splash_screen
2023-12-04 19:58:30,713 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
[... 107 lines cut ...]
  File "/home/umeko/KlipperScreen/panels/network.py", line 157, in add_network
    if network_id != -1 or netinfo['connected']:
NoneType: None
2023-12-04 20:00:44,897 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'connected'
  File "/home/umeko/KlipperScreen/panels/network.py", line 289, in scan_callback
    self.add_network(net, False)

  File "/home/umeko/KlipperScreen/panels/network.py", line 157, in add_network
    if network_id != -1 or netinfo['connected']:
NoneType: None
2023-12-04 20:00:44,910 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'connected'
  File "/home/umeko/KlipperScreen/panels/network.py", line 289, in scan_callback
    self.add_network(net, False)

  File "/home/umeko/KlipperScreen/panels/network.py", line 157, in add_network
    if network_id != -1 or netinfo['connected']:
NoneType: None
2023-12-04 20:00:44,924 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'connected'
  File "/home/umeko/KlipperScreen/panels/network.py", line 289, in scan_callback
    self.add_network(net, False)

  File "/home/umeko/KlipperScreen/panels/network.py", line 157, in add_network
    if network_id != -1 or netinfo['connected']:
NoneType: None
```

alfrix, 2024-03-17:
you should test if this happens on #1269, because i will not try to fix the old backend that i consider deprecated, the only reason that i haven't changed it it's because i'm giving people on buster some extra time to update.

umeiko, 2024-03-20:
I have tried your new branch, It seems good, but rescan function seems not working good.
```
024-03-20 11:41:25,266 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen >  > network
2024-03-20 11:41:28,595 [functions.py:logging_exception_handler()] - Uncaught exception <class 'sdbus_block.networkmanager.exceptions.NmDeviceNotAllowedError'>: Scanning not allowed immediately following previous scan
  File "/home/umeko/KlipperScreen/panels/network.py", line 309, in reload_networks
    self.sdbus_nm.rescan()

  File "/home/umeko/KlipperScreen/ks_includes/sdbus_nm.py", line 227, in rescan
    return self.wlan_device.request_scan({})

  File "/home/umeko/.KlipperScreen-env/lib/python3.8/site-packages/sdbus/dbus_proxy_sync_method.py", line 98, in __call__
    return self._call_dbus_sync(*rebuilt_args)

  File "/home/umeko/.KlipperScreen-env/lib/python3.8/site-packages/sdbus/dbus_proxy_sync_method.py", line 82, in _call_dbus_sync
    reply_message = self.interface._attached_bus.call(
NoneType: None
```

umeiko, 2024-03-20:
some time this, with very very long time scaning:
```2024-03-20 11:49:48,320 [menu.py:evaluate_enable()] - moonraker connected True
2024-03-20 11:49:48,779 [screen.py:show_panel()] - Reinitializing panel
2024-03-20 11:49:48,850 [network.py:__init__()] - Network interfaces: ['lo', 'ipa_lan0', 'rmnet_ipa0', 'wlan0']
2024-03-20 11:49:48,866 [network.py:__init__()] - Wireless interfaces: ['wlan0']
2024-03-20 11:49:48,876 [network.py:__init__()] - Primary interface: wlan0
2024-03-20 11:49:48,907 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen >  > network
2024-03-20 11:49:52,035 [functions.py:logging_exception_handler()] - Uncaught exception <class 'sdbus.dbus_exceptions.DbusUnknownMethodError'>: No such interface “org.freedesktop.DBus.Properties” on object at path /org/freedesktop/NetworkManager/AccessPoint/18
  File "/home/umeko/KlipperScreen/panels/network.py", line 75, in load_networks
    self.add_network(net['BSSID'])

  File "/home/umeko/KlipperScreen/panels/network.py", line 86, in add_network
    net = next(net for net in self.sdbus_nm.get_networks() if bssid == net['BSSID'])

  File "/home/umeko/KlipperScreen/ks_includes/sdbus_nm.py", line 149, in get_networks
    networks.extend(

  File "/home/umeko/KlipperScreen/ks_includes/sdbus_nm.py", line 161, in <genexpr>
    if ap.ssid

  File "/home/umeko/.KlipperScreen-env/lib/python3.8/site-packages/sdbus/dbus_proxy_sync_property.py", line 89, in __get__
    reply_message = obj._attached_bus. \
NoneType: None```

alfrix, 2024-03-20:
Since you switched to the branch manually you need to install the additional requirements
```
source ~/.KlipperScreen-env/bin/activate
pip -r ~/KlipperScreen/scripts/KlipperScreen-requirements.txt
sudo service KlipperScreen restart
deactivate
```
this is covered in the docs in the [manual update section](https://klipperscreen.github.io/KlipperScreen/Updating/#manual-update)

normally the update-manager takes care of this step

umeiko, 2024-03-21:
These logs shown afterwards I have done these requirements installation on klipperscreen venv.

alfrix, 2024-04-01:
i did some changes to the new backend branch that should resolve theses issues, except the slowness to scan, that may be tied to the hw, but it should not let you request another scan if the previous did not finish which seems to be the main error.

alfrix, 2024-05-24:
new backend merged

closed 2024-05-24 by alfrix