KlipperScreen/KlipperScreen #1319: [BUG] Recent update seems to have triggered an install of the network-manager component

dotorg, 2024-04-01:
### What happened?

Ran an update today of KlipperScreen, and as part of the update either KlipperScreen, or something it pulled in, appears to have installed network-manager automatically. It appears to have been added as a dependency a couple of months ago.

### What did you expect to happen instead?

Low-level system components not touched by an application update.

### How to reproduce this bug?

Update klipperscreen on a default OS install that does not use network-manager.

### Additional information:

This is an incredibly serious bug -- a fundamental, low-level OS component should *never* be blindly installed by a user-level application like KlipperScreen. 

On top of the security implications of it, network-manager starts doing MAC randomization by default, which will break any environment where people are using assigned DHCP addresses or using VLANs.  Its doubly-bad because most Klipper printers are using Moonraker's updates and are likely pulling this update automatically. 

Re-removing the network-manager package using apt fixes the problems its installation causes. 

In addition, it looks like it is also pulling in polkit, which is even more inappropriate to silently install, because of the *massive* security implications of being installed and potentially not properly configured for the target system. 

### Log output

```shell
n/a
```

alfrix, 2024-04-01:
network-manager is needed for the selection of the network in the wifi panel, as no other methods will be supported shortly, i agree that it could be optional and just showing a message with something like  `network-manager is needed for this function do: "sudo apt-install network-manager"`
polkit is needed for the ability to scan networks but also reboot/poweroff on systems without moonraker, moonraker itself uses almost the same permissions, (and some additional ones)

dotorg, 2024-04-01:
Yeah, to be clear, the problem is not that KlipperScreen is using network-manager for configuring WiFi, but that it installed it on a pre-existing system that didn't use it, which broke networking. 

Same with installing polkit. Its fine to require it, I suppose (although I think that's debatable) but installing a core security component in a system that is potentially being used by inexperienced Linux users is irresponsible. 

But we had a pretty serious production outage as a result of printers changing IP addresses with the update. It wasn't something noticed during upgrade testing because it never would've crossed anyone's mind that an app update would silently and fundamentally change networking configuration at the OS level.

alfrix, 2024-04-01:
for future reference, system-dependencies.json controls what's installed by moonraker in an update

dotorg, 2024-04-01:
Yeah, I'm running in a forked version because I needed the extrusion amounts to be floats instead of ints in extrude.py, and I've removed it from there in by forked version. (That's what caused us to run into the issue -- forking pulled the latest release, which caused us to pull that change and break all the printers.)

IMO, if the wireless support requires network-manager, the whole panel should be conditional on its install rather than forcing its install. Same with anything that may be requiring polkit. Shutdown/reboot don't need polkit to work properly, particularly given that it is not included on most trimmed-down Linux distributions, and its installation on older versions of Raspbian is known to break the systems. Anything else that needs it should be conditional.