KlipperScreen/KlipperScreen #1320: [BUG] <title>Missing Button for Multiple Instances

sweenbeen, 2024-04-02:
### What happened?

Recently did a full update on my BTT Pad7 - suddenly my Klipperscreen is not showing the button to my other instances - anyone have any tips? I still have a KlipperScreen.conf in my LS -L

### What did you expect to happen instead?

Expected to see button for instances - though I can still control each instance from my PC

### How to reproduce this bug?

Running KlipperScreen v0.3.9-72-ge62b0ac4

[logs-20240402-165836.zip](https://github.com/KlipperScreen/KlipperScreen/files/14843762/logs-20240402-165836.zip)

[image]

### Log output

```shell
Logs attached above
```

sweenbeen, 2024-04-02:
[image]

alfrix, 2024-04-02:
tat configuration isn't shared between the interfaces

your logs include klipper moonraker and mainsail but not klipperscreen log
please provide it to continue
https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/

sweenbeen, 2024-04-03:
Following the Sys Log /First Steps - KlipperScreen.log was not avail in the mainsail interface. So I had to pull up PuTTY to pull the info - though it does not look to be complete.  I will pull the SD card to verify locally.

biqu@BTT-Pad7:~$ systemctl status KlipperScreen
● KlipperScreen.service - KlipperScreen
     Loaded: loaded (/etc/systemd/system/KlipperScreen.service; enabled; vendor>
     Active: active (running) since Tue 2024-04-02 16:46:19 EDT; 15h ago
   Main PID: 430 (KlipperScreen-s)
      Tasks: 27 (limit: 1017)
     Memory: 141.4M
        CPU: 20min 34.890s
     CGroup: /system.slice/KlipperScreen.service
             ├─ 430 /bin/sh /home/biqu/KlipperScreen/scripts/KlipperScreen-star>
             ├─ 536 /usr/bin/xinit /home/biqu/.KlipperScreen-env/bin/python /ho>
             ├─ 657 /usr/lib/xorg/Xorg :0
             └─1138 /home/biqu/.KlipperScreen-env/bin/python /home/biqu/Klipper>

Apr 03 07:38:37 BTT-Pad7 KlipperScreen-start.sh[1138]: 2024-04-03 07:38:37,551 >
Apr 03 07:38:37 BTT-Pad7 KlipperScreen-start.sh[1138]: 2024-04-03 07:38:37,595 >
Warning: journal has been rotated since unit was started, output may be incompl>
lines 1-16/16 (END)...skipping...
● KlipperScreen.service - KlipperScreen
     Loaded: loaded (/etc/systemd/system/KlipperScreen.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2024-04-02 16:46:19 EDT; 15h ago
   Main PID: 430 (KlipperScreen-s)
      Tasks: 27 (limit: 1017)
     Memory: 141.4M
        CPU: 20min 34.890s
     CGroup: /system.slice/KlipperScreen.service
             ├─ 430 /bin/sh /home/biqu/KlipperScreen/scripts/KlipperScreen-start.sh
             ├─ 536 /usr/bin/xinit /home/biqu/.KlipperScreen-env/bin/python /home/biqu/KlipperScreen/screen.py
             ├─ 657 /usr/lib/xorg/Xorg :0
             └─1138 /home/biqu/.KlipperScreen-env/bin/python /home/biqu/KlipperScreen/screen.py

Apr 03 07:38:37 BTT-Pad7 KlipperScreen-start.sh[1138]: 2024-04-03 07:38:37,551 [screen.py:close_screensaver()] - Closing Screensaver
Apr 03 07:38:37 BTT-Pad7 KlipperScreen-start.sh[1138]: 2024-04-03 07:38:37,595 [screen.py:wake_screen()] - Screen wake up
Warning: journal has been rotated since unit was started, output may be incomplete.
~
~
~

biqu@BTT-Pad7:~$ journalctl -xe -u KlipperScreen
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
-- Journal begins at Wed 2024-04-03 06:59:29 EDT, ends at Wed 2024-04-03 07:49:2
Apr 03 07:38:37 BTT-Pad7 KlipperScreen-start.sh[1138]: 2024-04-03 07:38:37,551 [
Apr 03 07:38:37 BTT-Pad7 KlipperScreen-start.sh[1138]: 2024-04-03 07:38:37,595 [

biqu@BTT-Pad7:~$ cat /var/log/xorg.0.log
cat: /var/log/xorg.0.log: No such file or directory

alfrix, 2024-04-03:
the log provided is still not valid, it doesn't contain anything, grab the log file as explained in the docs
https://klipperscreen.readthedocs.io/en/latest/Troubleshooting

sweenbeen, 2024-04-03:
Sorry about that - I ended up having to install WinSCP to snag it.

[KlipperScreen.log](https://github.com/KlipperScreen/KlipperScreen/files/14856394/KlipperScreen.log)
[Xorg.0.log](https://github.com/KlipperScreen/KlipperScreen/files/14856395/Xorg.0.log)

alfrix, 2024-04-03:
```
2024-04-03 08:31:21,893 [config.py:__init__()] - Config path location: /home/biqu/printer_data/config/KlipperScreen.conf
2024-04-03 08:31:21,932 [config.py:log_config()] -  ===== Config File =====

=======================
2024-04-03 08:31:21,937 [config.py:__init__()] - ====== Saved Def ======

[main]
24htime = False
show_heater_power = True
show_scroll_steppers = True

=======================
2024-04-03 08:31:21,939 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
```
the switch printer button doesn't appear because you haven't configured the printers in the configuration file of klipperscreen, as said the interface configuration is not shared

sweenbeen, 2024-04-05:
Thanks for the reply - but it is in fact configured in the KlipperScreen.conf

[main]
default_printer: Green3v2

[printer Blue3]
#bottom usb - Ender5Plus
moonraker_host: localhost
moonraker_port: 7125
moonraker_api_key: False

[printer Green3v2]
#top usb
moonraker_host: localhost
moonraker_port: 7126
moonraker_api_key: False

[printer Red3Pro]
#side usb - Ender3Max
moonraker_host: localhost
moonraker_port: 7127
moonraker_api_key: False

[printer Sidewinder]
#side usb hub - None
moonraker_host: localhost
moonraker_port: 7128
moonraker_api_key: False

[KlipperScreen.conf.txt](https://github.com/KlipperScreen/KlipperScreen/files/14888813/KlipperScreen.conf.txt)

alfrix, 2024-04-05:
no it's not, because it's not reading the config that you have shown, it's reading what i've posted, where is you config file? i mean the absolute path to the file

sweenbeen, 2024-04-05:
10-4 - got it by nav to "/home/biqu/printer_data/config/KlipperScreen.conf" and updating that config file.

Previously the conf was added by "sudo cat>KlipperScreen.conf" which i thought built it in the root.

closed 2024-04-05 by sweenbeen

alfrix, 2024-04-05:
you can always create symbolic links to access the configuration file from the browser of different printers
for example:
```
 ln -s /home/biqu/printer_data/config/KlipperScreen.conf /home/biqu/printer_1_data/config/KlipperScreen.conf
 ```
 remember that KlipperScreen only allows 1 configuration for all the printers, the menu system will filter menus by machine capabilities

sweenbeen, 2024-04-05:
Truth be had, I am still very new to ssh and python and pi work. so i will have to do some research on your symbolic link item. I used Kuiah to install 4 instances - so your comment makes sense but also corrects what i was thinking it was doing