KlipperScreen/KlipperScreen #1322: No documentation for creating menus in klipper screen

John-SC077, 2024-04-04:
### Requested feature

I spent 4 hours researching the correct syntax for the menu items in the klipper screen config file. There are dozens of samples from three or more years ago on the web and none of them work because apparently the syntax was changed at some recent time.  Is there a document that describes how to create a menu item and what options can or must be provided for each menu item?  For instance, many samples on the web have a "gcode:" option with a line or more of gcode commands after it.  But the klipper screen application complains about that during startup and will not start with that in the file.  It now has to be on a "params:" line.

### Is your feature request related to a problem?

Klipper Screen does not have an option to use the bed_screws_adjust macro, so leveling a bed on a printer that has no automated measuring devices is not possible directly.  And creating a menu structure to add this ability requires knowing how to format the instructions in the config file.

### Describe alternatives you have considered

I managed to find enough samples that I could eventually create this (tiny) set of menu items, but I am not certain why it works or how to enhance it in the future if I need to.

[menu __main __leveling]
name: Leveling
icon: home

[menu __main __leveling Home]
name: Home All
icon: home
method: printer.gcode.script
params: {"script":"G28"}

[menu __main __leveling Bed_Adjust]
name: Bed Adjust
icon: Adjust
method: printer.gcode.script
params: {"script":"Bed_Screws_Adjust"}

[menu __main __leveling Accept]
method: printer.gcode.script
name: Accept
params: {"script":"ACCEPT"}

[menu __main __leveling Adjust]
method: printer.gcode.script
name: Adjust
params: {"script":"ADJUSTED"}

[menu __main __leveling Abort]
method: printer.gcode.script
name: Abort
params: {"script":"G1 Z10"}

### Additional information

I understand that documentation is a hard thing to create, I am willing to help but I do not want to create a partial or wrong document.  If someone can point me in the direction of some set of data that would let me create one, and provide reviews of that finished product I would be happy to write it.

John-SC077, 2024-04-06:
Thanks for the quick response.  My confusion comes in large part from the link https://www.klipper3d.org/Config_Reference.html?h=menu which I understand now is for the printer.cfg file, while I was creating a klipperscreen configuration file.  And now from your response I understand they are different, the menu parsing was reinvented.  So somewhere in the klipperscreen source there is a parser that interprets that configuration file and renders and executes it.  I guess I will go searching for that.  But the information I was looking for is basically what is in the link I listed above but for the klipperscreen instead of the klipper main config, with all (or most) of the available syntax options.  I am not sure if the first block of code in that link shows the complete set of syntax - if it does that is fine, but specific questions I have are what are valid methods,  and what params are valid, and how are they formatted?

If that is not available I will start hunting code, if you can point me to a subset of where to look I would appreciate that help;

alfrix, 2024-04-07:
methods are from moonraker as said in the docs, i will add a link to the moonraker docs, usually only printer.gcode.script is used but there are others like `printer.print.cancel` or `printer.firmware_restart` some methods need paramas other don't

> first block of code in that link shows the complete set of syntax

only confirm is missing, i will add it and consider this resolved, if you want to add or modify the docs please do a PR or at least open an issue with the proposed changes

Thank you

commit 11cebe8c (2024-04-07): docs: update menu close #1322
closed 2024-04-07 by alfrix