KlipperScreen/KlipperScreen #1328: [BUG] Bed Level UI Fails to Render Properly

cneshi, 2024-04-11:
### What happened?

When I enter the `More->Bed Level` Menu, the arrows are not rendered properly, and pressing the arrow buttons (that are present) do not match the `bed_screw_adjust` points in the Klipper config. Using `screw_rotation` did not seem to fix the issue. 

I believe this issue is occuring because I have the Y axis inverted (the print bed is upside down). 

### What did you expect to happen instead?

I believe there should be some code in `bed_level.py` that checks y axis inversion in the `klipperscreen.conf` and swap out y-max and y-min when attempting to match the `fr, fl, fm, bl, br, bm` points. We did this manually in the code and it fixed the issue for us.

### How to reproduce this bug?

The physical bed has dimensions of 180x180

In `klipper config` put in the following configs: 
```
[bed_screws]
screw1: 5,170
screw1_name: fl
screw2: 180,170
screw2_name: fr
screw3: 90,20
screw3_name: bm
```

In `klipperscreen.conf` put in the following configs:
`screw_positions: fl, fr, bm`

Restart Klipperscreen and enter the `More->Bed Level` menu

### Additional information:

The relavant code is in `bed_level.py` line 137 to 146

### Log output

=======================
2024-04-10 23:08:05,696 [config.py:__init__()] - Configured printers: [
  {
    "Positron_V3": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2024-04-10 23:08:05,769 [config.py:create_translations()] - Selected lang: None OS lang: en_GB
2024-04-10 23:08:05,769 [config.py:install_language()] - Using lang en
2024-04-10 23:08:05,780 [screen.py:__init__()] - Monitors: 1 using number: 0
2024-04-10 23:08:05,781 [screen.py:__init__()] - Wayland: False Display name: :0
2024-04-10 23:08:05,783 [screen.py:__init__()] - Screen resolution: 640x480
2024-04-10 23:08:05,784 [KlippyGtk.py:__init__()] - Font size: 14.9 (medium)
2024-04-10 23:08:06,077 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 3600
2024-04-10 23:08:06,086 [screen.py:set_screenblanking_timeout()] - Using DPMS
2024-04-10 23:08:06,094 [screen.py:initial_connection()] - Default printer: None
2024-04-10 23:08:06,118 [screen.py:connect_printer()] - Connecting to printer: Positron_V3
2024-04-10 23:08:06,118 [screen.py:_load_panel()] - Loading panel: splash_screen
2024-04-10 23:08:06,152 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2024-04-10 23:08:06,160 [KlippyWebsocket.py:connect()] - Attempting to connect
2024-04-10 23:08:06,160 [KlippyRest.py:_do_request()] - Sending get to http://127.0.0.1:7125/server/info
2024-04-10 23:08:06,174 [KlippyRest.py:_do_request()] - Sending get to http://127.0.0.1:7125/access/oneshot_token
2024-04-10 23:08:06,185 [KlippyWebsocket.py:connect()] - Starting websocket thread
2024-04-10 23:08:06,192 [_logging.py:info()] - Websocket connected
2024-04-10 23:08:06,193 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2024-04-10 23:08:06,197 [KlippyRest.py:_do_request()] - Sending get to http://127.0.0.1:7125/server/info
2024-04-10 23:08:06,209 [KlippyRest.py:_do_request()] - Sending get to http://127.0.0.1:7125/server/webcams/list
2024-04-10 23:08:06,219 [printer.py:configure_cameras()] - Cameras: []
2024-04-10 23:08:06,219 [screen.py:init_klipper()] - Moonraker info {'klippy_connected': True, 'klippy_state': 'ready', 'components': ['secrets', 'template', 'klippy_connection', 'jsonrpc', 'internal_transport', 'application', 'websockets', 'dbus_manager', 'database', 'file_manager', 'authorization', 'klippy_apis', 'shell_command', 'machine', 'data_store', 'proc_stats', 'job_state', 'job_queue', 'http_client', 'announcements', 'webcam', 'extensions', 'octoprint_compat', 'history', 'update_manager'], 'failed_components': [], 'registered_directories': ['config', 'logs', 'gcodes', 'config_examples', 'docs'], 'warnings': [], 'websocket_count': 4, 'moonraker_version': 'v0.8.0-325-g99b97af', 'missing_klippy_requirements': [], 'api_version': [1, 4, 0], 'api_version_string': '1.4.0'}
2024-04-10 23:08:06,220 [KlippyRest.py:_do_request()] - Sending get to http://127.0.0.1:7125/printer/info
2024-04-10 23:08:06,233 [KlippyRest.py:_do_request()] - Sending get to http://127.0.0.1:7125/printer/objects/query?configfile
[... 41 lines cut ...]
2024-04-10 23:08:18,748 [bed_level.py:_get_screws()] - screw3: 90,20
2024-04-10 23:08:18,748 [bed_level.py:_get_screws()] - screw3_name: bm
2024-04-10 23:08:18,748 [bed_level.py:__init__()] - bed_screws: [[90.0, 20.0], [5.0, 170.0], [180.0, 170.0]]
2024-04-10 23:08:18,748 [bed_level.py:__init__()] - Positions: ['fl', 'fr', 'bm']
2024-04-10 23:08:18,748 [bed_level.py:__init__()] - Rotation: 0
2024-04-10 23:08:18,748 [bed_level.py:__init__()] - X: {90.0, 180.0, 5.0}
Y: {170.0, 20.0}
2024-04-10 23:08:18,749 [bed_level.py:__init__()] - Using max_distance: 116 to fit: 3 screws.
2024-04-10 23:08:18,749 [bed_level.py:__init__()] - Screws not used: [[180.0, 170.0]]
2024-04-10 23:08:18,749 [bed_level.py:__init__()] - Using 2/3-screw locations
2024-04-10 23:08:18,783 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > more > bed_level
2024-04-10 23:08:22,658 [screen.py:_menu_go_back()] - #### Menu go back
2024-04-10 23:08:22,697 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > more
2024-04-10 23:08:22,708 [menu.py:arrangeMenuItems()] - X > more bedmesh
2024-04-10 23:08:22,712 [menu.py:arrangeMenuItems()] - X > more retraction
2024-04-10 23:08:22,724 [menu.py:arrangeMenuItems()] - X > more power
2024-04-10 23:08:22,727 [menu.py:arrangeMenuItems()] - X > more camera
2024-04-10 23:08:22,732 [menu.py:arrangeMenuItems()] - X > more input_shaper
2024-04-10 23:08:22,739 [menu.py:arrangeMenuItems()] - X > more spoolman
2024-04-10 23:08:23,169 [screen.py:_menu_go_back()] - #### Menu go back
2024-04-10 23:08:23,185 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2024-04-10 23:08:26,999 [screen.py:_load_panel()] - Loading panel: gcode_macros
2024-04-10 23:08:27,032 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > gcode_macros
2024-04-10 23:08:28,453 [screen.py:_menu_go_back()] - #### Menu go back
2024-04-10 23:08:28,483 [screen.py:attach_panel()] - Current panel hierarchy: main_menu

alfrix, 2024-04-11:
the issue you are having is not really a bug
based on the coordinates your config should be `screw_positions: bl, br, fm` 
the reason is that is the convention for Front-Left is 0,0 Back-Right Max-X and Max-Y.

the display would need to be then rotated using screw rotation but, you are right there is no invert support.
inversion based on config could be the used to rotate the interface visually
so to me this would be a FR of support inversion to show the bed screws (and probably mesh)

Furthermore i have researched a little about this printer, and it seems that it homes front left (Steve-Builds YT), so all the coordinates and everything should work normally without any special inversion, i don't understand why you are using inverted Y

cneshi, 2024-04-12:
The printhead does indeed home front left but the front left coordinates are 0,180(max y). The rear left is 0,0. As I mentioned, the whole system is upside down. Imagine a normal printer where the front left is 0,0 and flipping the bed using the back edge as the pivot. The left and right stay the same, but the front and back coordiates are inverted.

PR #1335 (2024-04-18): bed_level: support inversion and fix screws_rotation of 3/6 screws
deprecate screw positions, we will be auto-detecting from now on

support rotating the layout (previously only rotated the button actions) and did not work properly for 3 screw configs.

inverting axis in KS will invert the layout fixes #1328

since this changes a lot of stuff it needs more testing
closed 2024-04-18 by alfrix