KlipperScreen/KlipperScreen #1329: [BUG] Loading panel: camera : cannot allocate memory in static TLS block

umeiko, 2024-04-12:
### What happened?

When I click camera logo, The error shows.

### What did you expect to happen instead?

show the camera panel.

### How to reproduce this bug?

I'm using Ubuntu-20.04 based on arm64 Linux devices. When I click camera logo, The error shows.

### Additional information:

I'm using Ubuntu 20.04 based on arm64 Linux devices. 

### Log output

```bash
2024-04-12 21:39:59,501 [screen.py:_load_panel()] - Loading panel: camera
2024-04-12 21:40:00,391 [screen.py:show_error_modal()] - Showing error modal: Unable to load panel camera /lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block

Traceback (most recent call last):
  File "/home/umeko/KlipperScreen/screen.py", line 321, in show_panel
    self.panels[panel_name] = self._load_panel(panel).Panel(self, title, **kwargs)
  File "/home/umeko/KlipperScreen/screen.py", line 308, in _load_panel
    return import_module(f"panels.{panel}")
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/umeko/KlipperScreen/panels/camera.py", line 1, in <module>
    import mpv
  File "/home/umeko/.KlipperScreen-env/lib/python3.8/site-packages/mpv.py", line 53, in <module>
    backend = CDLL(sofile)
  File "/usr/lib/python3.8/ctypes/__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block

2024-04-12 21:40:00,425 [functions.py:get_software_version()] - Error getting git version: b"fatal: detected dubious ownership in repository at '/home/umeko/KlipperScreen'\nTo add an exception for this directory, call:\n\n\tgit config --global --add safe.directory /home/umeko/KlipperScreen\n"
```

alfrix, 2024-04-13:
The issue comes from `import mpv` so the issue is not from this repi would check the version of python-mpv
```
source ~/.KlipperScreen-env/bin/activate
pip freeze | grep python-mpv
``` 
if its `python-mpv==1.0.5` i would maybe raise an issue in their repo
if its older you can try to update, if the update doesn't work chances are that you need to update to a newer/different distro.

alfrix, 2024-04-13:
i can't do anything to fix this since it comes from dependencies

closed 2024-04-13 by alfrix

umeiko, 2024-04-24:
yep, seems `python-mpv==1.0.5
`
```bash
(.KlipperScreen-env) umeko@umeko-hm2:~$ pifreeze | grep python-mpv
python-mpv==1.0.5
```