KlipperScreen/KlipperScreen #1336: Power ON button on the home screen

blonestar, 2024-04-16:
### Requested feature

Power ON button would be nice to have within click on the main/home screen for relay controlled printers.

[image]
[image]

### Is your feature request related to a problem?

I find just a bit hard to turn on my printer from the KlipperScreen as I'm using relay to control it.
On the other hand it is quite easy to do that from the Mainsail.
[image]

### Describe alternatives you have considered

Need to open Menu > Power to get to the Relay/Power controls

[image]

### Additional information

Relays are controlled  through the Moonraker and GPIOs on the Pi (BTT Pi in this case)

[image]

mjfsch, 2024-04-16:
This already exists - however it is sometimes a little tricky to get working 

The main thing to get it working, is the power device and the printer need to have the same name, currently your power device is called "Printer" but your printer is called "Trident"

https://github.com/KlipperScreen/KlipperScreen/issues/1241

blonestar, 2024-04-16:
Just added this line to the klipperscreen.conf and it's working.
`power_devices: Printer`
Thank you, 
will close the issue

[image]

closed 2024-04-16 by blonestar