KlipperScreen/KlipperScreen #1358: [BUG] Updated and KlipperScreen no longer works

onkel645, 2024-05-06:
### What happened?

Hello.
Sorry for my bad english...
I did an update and then Klipperscreen stopped working. Now Klipperscreen can`t connect to Moonraker (is on the display).
I reinstalled Moonraker and Klipperscreen but it doesn't work.

KlipperScreen
v0.4.0-49-g32d0af46

moonraker
v0.8.0-354-gba942855

mainsail
v2.11.2

mainsail-config
v1.2.1-0-ge57810d5

### What did you expect to happen instead?

Restarting the printer doesn't help

### How to reproduce this bug?

Restarting the printer doesn't help

### Log output

[KlipperScreen.log](https://github.com/KlipperScreen/KlipperScreen/files/15223610/KlipperScreen.log)
[moonraker.log](https://github.com/KlipperScreen/KlipperScreen/files/15223612/moonraker.log)

alfrix, 2024-05-06:
nothing to do with any update, you have a malformed configuration, this is how a printer section is supposed to look
```
[printer MyPrinter]
moonraker_host: 127.0.0.1
moonraker_port: 7125
moonraker_api_key: aiofjaofiaiofaifoanifanofan
 ```

the section header is in the brackets [] and it's not optional

onkel645, 2024-05-06:
Thanks, it works again!
Perfect👍

closed 2024-05-06 by onkel645