KlipperScreen/KlipperScreen #1368: [BUG] BTT TFT35 SPI screen not working on BTT CB2

bddanford, 2024-05-16:
### What happened?

while using the new BTT CB2, I get no KlipperScreen popping up. I get the console login prompt on the screen, but no Xserver and no KlipperScreen showing up. Have removed and reinstalled KS with Kiauh and the KS install and no luck. I think it has something to do with the Xserver config for the OS that BTT ships for the CB2.

Welcome to Armbian 23.05.0-trunk Bullseye with Linux 4.19.232-bigtree-cb2

### What did you expect to happen instead?

The printer to boot and KS come up. Worked fine before I swapped out the CB1/old OS with the new CB2/new OS. 

### How to reproduce this bug?

Remove/Reinstall KS, and reboot

### Log output

I removed the KS.log file to get a clean one, but it never recreated it. Here is the output for the KS-start.sh script. 

biqu@v02:~$ cd KlipperScreen/
biqu@v02:~/KlipperScreen$ cd scripts/
./KlipperScreen-start.sh: 14: [[: not found
Running KlipperScreen on X in display :0 by default

X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Debian
Current Operating System: Linux v02 4.19.232-bigtree-cb2 #2 SMP Mon Apr 8 18:40:16 CST 2024 aarch64
Kernel command line: root=UUID=269e1e57-7d4c-484f-b690-41c2602c73d2 rootwait rootfstype=ext4 splash plymouth.ignore-serial-consoles console=tty1 consoleblank=0 loglevel=1 ubootpart=0173df3f-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u   cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
Build Date: 10 April 2024  08:59:35AM
xorg-server 2:1.20.11-1+deb11u13 (https://www.debian.org/support) 
Current version of pixman: 0.40.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Thu May 16 12:25:25 2024
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE) 
Fatal server error:
(EE) no screens found(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.

biqu@v02:~/KlipperScreen/scripts$ tail -n 50 /var/log/Xorg.0.log
[   567.669] 	ABI class: X.Org Video Driver, version 24.1
[   567.669] (II) LoadModule: "fbdev"
[   567.669] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[   567.670] (II) Module fbdev: vendor="X.Org Foundation"
[   567.670] 	compiled for 1.20.0, module version = 0.5.0
[   567.670] 	Module class: X.Org Video Driver
[   567.670] 	ABI class: X.Org Video Driver, version 24.0
[   567.670] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[   567.670] (II) FBDEV: driver for framebuffer: fbdev
[   567.672] (II) modeset(0): using drv /dev/dri/card0
[   567.672] (WW) Falling back to old probe method for fbdev
[   567.672] (II) Loading sub module "fbdevhw"
[   567.672] (II) LoadModule: "fbdevhw"
[   567.672] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[   567.673] (II) Module fbdevhw: vendor="X.Org Foundation"
[   567.673] 	compiled for 1.20.11, module version = 0.0.2
[   567.673] 	ABI class: X.Org Video Driver, version 24.1
[   567.673] (WW) VGA arbiter: cannot open kernel arbiter, no multi-card support
[   567.674] (II) modeset(0): Creating default Display subsection in Screen section
	"Default Screen Section" for depth/fbbpp 24/32
[   567.674] (==) modeset(0): Depth 24, (==) framebuffer bpp 32
[   567.674] (==) modeset(0): RGB weight 888
[   567.674] (==) modeset(0): Default visual is TrueColor
[   567.674] (II) Loading sub module "glamoregl"
[   567.674] (II) LoadModule: "glamoregl"
[   567.674] (II) Loading /usr/lib/xorg/modules/libglamoregl.so
[   567.691] (II) Module glamoregl: vendor="X.Org Foundation"
[   567.692] 	compiled for 1.20.11, module version = 1.0.1
[   567.692] 	ABI class: X.Org ANSI C Emulation, version 0.4
[   568.389] (II) modeset(0): Refusing to try glamor on llvmpipe
[   568.392] (EE) modeset(0): glamor initialization failed
[   568.392] (II) modeset(0): ShadowFB: preferred NO, enabled NO
[   568.393] (WW) modeset(0): No outputs definitely connected, trying again...
[   568.393] (WW) modeset(0): Unable to find connected outputs - setting 1024x768 initial framebuffer
[   568.393] (==) modeset(0): Using gamma correction (1.0, 1.0, 1.0)
[   568.393] (EE) modeset(0): No modes.
[   568.393] (II) UnloadModule: "modesetting"
[   568.393] (II) UnloadSubModule: "glamoregl"
[   568.393] (II) Unloading glamoregl
[   568.393] (EE) Screen(s) found, but none have a usable configuration.
[   568.393] (EE) 
Fatal server error:
[   568.393] (EE) no screens found(EE) 
[   568.393] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[   568.393] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[   568.393] (EE) 
[   568.395] (EE) Server terminated with error (1). Closing log file.

alfrix, 2024-05-24:
this is not a bug of klipperscreen, you need to ensure the screen is properly connected and configured with the correct drivers installed, unfortunately most screens and platforms differ in their implementation
ultimately the manufacturer should document how the screen should be installed
i only have raspberries so this platforms are unknown to me

technical50, 2024-05-27:
Same issue here Im not Screen not working anymore...

igl-x2, 2024-06-02:
try tapping the screen while having the Mainsail website open.
your screen probably works, it is just running at 0 brightness level.

its a bug with the btt/armbian software.
https://github.com/bigtreetech/CB2/issues/10

alfrix, 2024-07-14:
> try tapping the screen while having the Mainsail website open. your screen probably works, it is just running at 0 brightness level.
> 
> its a bug with the btt/armbian software. [bigtreetech/CB2#10](https://github.com/bigtreetech/CB2/issues/10)

unlikely, the driver is not configured properly and cannot start, this is not fixable by KS it needs the user needs to install the screen properly, BTT should be the ones providing support since it's their hw

closed 2024-07-14 by alfrix