KlipperScreen/KlipperScreen #1372: [FR] Better UI text and icon scaling on small screens

kb-elmo, 2024-05-23:
### Requested feature

Right now the UI scaling on small screens is unfortunately horrible.  
The icons and especially the text is **way** too small with tons of wasted padding space on buttons.

[image]

Changing the font size in the settings unfortunately doesn't really do anything for the status screens.  
Right now it seems that it just scales up the text in the menus and dialogues.

Since there is so much empty padding space around the text I thought it might be easy to change it in a custom theme.  
But unfortunately this just causes the entire UI to fall apart and parts of it disappearing (mainly the action bar at the bottom being pushed out of screen by the main content due to it's overly excessive padding).

Since most printers nowadays come with screens somewhere in the 3.5-5 inch range it would imho make sense to fit the UI to those displays.
Only very few people are going to have large touch tablets mounted to their printers.

### Is your feature request related to a problem?

Right now the UI is borderline useless on smaller (standard 3D printer) screens

### Describe alternatives you have considered

Right now KlipperScreen is pretty much without alternative when it comes to a proper touchscreen UI for Klipper based printers

### Additional information

I really really like KlipperScreen since it provides a very well thought out UI with all the information one would need.  
Only the bad scaling on smaller screens is a major downside right now.

I think making the UI scale properly for smaller screens would really benefit a lot of other people as well.

alfrix, 2024-05-24:
what i take from this and your image is that the main-menu and temperature have buttons too big in vertical mode and more space could be dedicated to the graph and heaters, i don't test vertical very much, i always use it horizontal

closed 2024-05-24 by alfrix

kb-elmo, 2024-05-24:
This would be nice as well, indeed.

But my main point is really the font size, especially on the temperatures.
It's so small that it's borderline unreadable for no reason that I can see.

There's more than enough space to make the font twice, or even three times as big without having to increase the size of the space it is currently taking up.
The rows could stay the exact same size, the font just has to be a lot bigger.

The entire UI scaling feels like it was meant to be looked at on a large computer monitor.
Not on a small 3D printer display.

kb-elmo, 2024-05-24:
With the new changes the UI falls apart completely when opening the temperature setting in the main screen.

[image]

It seems that the graph doesn't get hidden and the numpad is just displayed on top of it.
The action bar and the rest of the main content is also pushed out of screen.

alfrix, 2024-05-24:
i see, i'll fix that

> The entire UI scaling feels like it was meant to be looked at on a large computer monitor.
Not on a small 3D printer display.

not at all, it's carefully tuned for horizontal orientation, vertical has always been an after tought, i see if i can increase the font size a little more, but due to the arrangement text on vertical is smaller than the horizontal counterpart in all cases

kb-elmo, 2024-05-24:
> i see, i'll fix that
>
>
> not at all, it's carefully tuned for horizontal orientation, vertical has always been an after tought, i see if i can increase the font size a little more, but due to the [...]

It's understandable that the text would scale smaller on a smaller screen since it has to fit the width of the screen.
But looking at it right now there is so much empty padding space around the text that it could easily double in size making it much more easy to read.

And even when in horizontal mode, the text is still really hard to read on those smaller 3.5 inch screens.

I really appreciate your work though either way :)

And I might try to look more into it myself when I get the time in the next few days.
Maybe I can create a custom theme for smaller displays that somehow scales up the font a bit.

Jaizu, 2024-05-28:
I can understand his point, having a more reponsive design for smaller screens would be nice.
This is a very bad mockup of how to make more use of the space so smaller screens can have more readable stuff on the screen, I hope you understand the point here:
[image]

commit 3fee9cbb (2024-05-24): main_menu and temp: change layout to accomodate a bigger graph and list in vertical mode

close #1372