KlipperScreen/KlipperScreen #1373: [BUG] Internal error during connect: cannot import name signature

h3rm, 2024-05-24:
### What happened?

after todays update i get the error:
https://imgur.com/cBKVtbn

### What did you expect to happen instead?

normal start

### How to reproduce this bug?

update alle components via web interface

### Log output

Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 175, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 141, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 121, in load_object
    mod = importlib.import_module('extras.' + module_name)
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/home/pi/klipper/klippy/extras/autotune_tmc.py", line 3, in <module>
    from inspect import signature
ImportError: cannot import name signature
webhooks client 4119131600: New connection
webhooks client 4119131600: Client info {'program': 'Moonraker', 'version': 'v0.8.0-359-g73df63d'}

alfrix, 2024-05-24:
this is not related to the KlipperScreen interface closing

it seems that it has been caused by your extension/plugin not being compatible with python 2.7, i suggest installing klipper with python 3.x

closed 2024-05-24 by alfrix

h3rm, 2024-05-24:
Maybe https://github.com/andrewmcgr/klipper_tmc_autotune/issues/174#issuecomment-2129454477 .. sry and thx