KlipperScreen/KlipperScreen #1377: [BUG] "Nothing Selected" error when selecting temperature preset

Turge08, 2024-05-26:
### What happened?

When I press one of the presets on one of my many printers with the latest updates, the message "Nothing Selected" appears and the temperatures aren't set.

### What did you expect to happen instead?

The preset should configure the proper temps for the hot end and bed

### How to reproduce this bug?

Update KlipperScreen and press one of the temperature presets

### Log output

2024-05-25 23:49:58,729 [screen.py:_load_panel()] - Loading panel: temperature
2024-05-25 23:49:58,814 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > temperature
2024-05-25 23:49:59,800 [screen.py:show_popup_message()] - error: Nothing selected

alfrix, 2024-05-26:
this is a bug in the sense that it doesn't pre-selects, but you can select what you want!

closed 2024-05-26 by alfrix

mendeste, 2024-05-26:
Closed but not completed still has bug
Yes pre-select working
But it opens fine tunes screen first select - +
Need to tap that to get the filament option

Turge08, 2024-05-26:
I'm really not clear on what the new direction is. The user is expected to select extruder and heater bed prior to tapping on the preset which are already defined to configure them? This is non-intuitive and now requires 3 taps to preheat the printer.

Hopefully this will either be reconsidered or an option to toggle this feature will be provided. 

Until then, I'll look at using gcode to restore this functionality.

alfrix, 2024-05-26:
> Closed but not completed still has bug Yes pre-select working But it opens fine tunes screen first select - + Need to tap that to get the filament option

oops, will fix this

> Hopefully this will either be reconsidered or an option to toggle this feature will be provided.

relax, already fixed the pre-select, but you should know that you can heat or cooldown selectively (not if you use gcode)

Turge08, 2024-05-26:
Sorry, I thought you just closed the issue.

mendeste, 2024-05-26:
Thank you all fixed back to normal 
Cheers

commit 95404269 (2024-05-26): temperature: fix pre-selection close #1377