KlipperScreen/KlipperScreen #1379: [BUG] <TFT35 Screen SPI connection's not working>

technical50, 2024-05-27:
### What happened?

After Upgrade Klipper Screen

### What did you expect to happen instead?

Klipper Screen flashes blackscreen - console again again

### How to reproduce this bug?

Upgrade issue

### Additional information:

Mks robin tft 35 using coonection maded my github
https://github.com/technical50/TFT35V1.0-Klipper-screen

### Log output

TFT35 SPI Connections issued after update My Screen not working anymore...

Im using Two Trees Sapphire Plus original Mks Robin TFT35 screen with SPI converter.
Details  you may see my github
https://github.com/technical50/TFT35V1.0-Klipper-screen

Not only me some people has issiues on facebook groups.
[image]

`[ 11516.111] 
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[ 11516.111] Build Operating System: Linux 5.4.0-109-generic armv8l Raspbian
[ 11516.111] Current Operating System: Linux mainsail2 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l
[ 11516.111] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i smsc95xx.macaddr=DC:A6:32:C0:B9:93 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=1cd7312d-02 rootfstype=ext4 fsck.repair=yes rootwait
[ 11516.111] Build Date: 04 April 2023  07:50:56AM
[ 11516.111] xorg-server 2:1.20.4-1+rpt4+deb10u9 (https://www.debian.org/support) 
[ 11516.111] Current version of pixman: 0.36.0
[ 11516.111] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[ 11516.111] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[ 11516.112] (==) Log file: "/var/log/Xorg.0.log", Time: Mon May 27 21:38:15 2024
[ 11516.112] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[ 11516.113] (==) No Layout section.  Using the first Screen section.
[ 11516.113] (==) No screen section available. Using defaults.
[ 11516.113] (**) |-->Screen "Default Screen Section" (0)
[ 11516.113] (**) |   |-->Monitor "<default monitor>"
[ 11516.113] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[ 11516.114] (**) |   |-->Device "Allwinner A10/A13 FBDEV"
[ 11516.114] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[ 11516.114] (==) Automatically adding devices
[ 11516.114] (==) Automatically enabling devices
[ 11516.114] (==) Automatically adding GPU devices
[ 11516.114] (==) Max clients allowed: 256, resource mask: 0x1fffff
[ 11516.114] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[ 11516.114] 	Entry deleted from font path.
[ 11516.114] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[ 11516.114] 	Entry deleted from font path.
[ 11516.114] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[ 11516.114] 	Entry deleted from font path.
[ 11516.114] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[ 11516.114] 	Entry deleted from font path.
[ 11516.114] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[ 11516.114] 	Entry deleted from font path.
[ 11516.114] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[ 11516.114] 	Entry deleted from font path.
[ 11516.114] (==) FontPath set to:
[... 203 lines cut ...]
[ 11516.663] (**) Option "xkb_model" "pc105"
[ 11516.663] (**) Option "xkb_layout" "us"
[ 11516.663] (**) vc4: Applying InputClass "evdev keyboard catchall"
[ 11516.663] (**) vc4: Applying InputClass "libinput keyboard catchall"
[ 11516.663] (II) Using input driver 'libinput' for 'vc4'
[ 11516.663] (**) vc4: always reports core events
[ 11516.663] (**) Option "Device" "/dev/input/event2"
[ 11516.663] (**) Option "_source" "_driver/libinput"
[ 11516.663] (II) libinput: vc4: is a virtual subdevice
[ 11516.663] (**) Option "config_info" "udev:/sys/devices/platform/soc/fef05700.hdmi/rc/rc1/input8/event2"
[ 11516.664] (II) XINPUT: Adding extended input device "vc4" (type: KEYBOARD, id 11)
[ 11516.664] (**) Option "xkb_model" "pc105"
[ 11516.664] (**) Option "xkb_layout" "us"
[ 11516.721] (II) event0  - USB 2.0 Camera: USB Camera: device removed
[ 11516.771] (II) event3  - ADS7846 Touchscreen: device removed
[ 11516.821] (II) event1  - vc4: device removed
[ 11516.880] (II) event2  - vc4: device removed
[ 11516.923] (II) UnloadModule: "libinput"
[ 11516.923] (II) UnloadModule: "libinput"
[ 11516.923] (II) UnloadModule: "libinput"
[ 11516.923] (II) UnloadModule: "libinput"
[ 11516.924] (II) UnloadModule: "libinput"
[ 11516.925] (II) UnloadModule: "libinput"
[ 11516.935] (II) Server terminated successfully (0). Closing log file.
`

technical50, 2024-05-27:
working one

`X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[    27.028] Build Operating System: Linux 5.4.0-109-generic armv8l Raspbian
[    27.028] Current Operating System: Linux mainsail2 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l
[    27.028] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i smsc95xx.macaddr=DC:A6:32:C0:B9:93 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=1cd7312d-02 rootfstype=ext4 fsck.repair=yes rootwait
[    27.028] Build Date: 04 April 2023  07:50:56AM
[    27.029] xorg-server 2:1.20.4-1+rpt4+deb10u9 (https://www.debian.org/support) 
[    27.029] Current version of pixman: 0.36.0
[    27.029] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    27.029] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    27.029] (==) Log file: "/var/log/Xorg.0.log", Time: Mon May 27 22:30:47 2024
[    27.063] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    27.093] (==) No Layout section.  Using the first Screen section.
[    27.093] (==) No screen section available. Using defaults.
[    27.093] (**) |-->Screen "Default Screen Section" (0)
[    27.093] (**) |   |-->Monitor "<default monitor>"
[    27.098] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[    27.098] (**) |   |-->Device "Allwinner A10/A13 FBDEV"
[    27.098] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    27.098] (==) Automatically adding devices
[    27.098] (==) Automatically enabling devices
[    27.098] (==) Automatically adding GPU devices
[    27.104] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    27.122] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    27.122] 	Entry deleted from font path.
[    27.122] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    27.122] 	Entry deleted from font path.
[    27.122] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    27.122] 	Entry deleted from font path.
[    27.122] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    27.122] 	Entry deleted from font path.
[    27.122] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    27.122] 	Entry deleted from font path.
[    27.122] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    27.122] 	Entry deleted from font path.
[    27.123] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	built-ins
[    27.123] (==) ModulePath set to "/usr/lib/xorg/modules"
[    27.123] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    27.123] (II) Loader magic: 0x1fdf80
[    27.123] (II) Module ABI versions:
[    27.123] 	X.Org ANSI C Emulation: 0.4
[    27.123] 	X.Org Video Driver: 24.0
[    27.123] 	X.Org XInput driver : 24.1
[    27.123] 	X.Org Server Extension : 10.0
[    27.124] (--) using VT number 2

[    27.124] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    27.127] (II) xfree86: Adding drm device (/dev/dri/card0)
[    27.128] (II) xfree86: Adding drm device (/dev/dri/card1)
[    27.129] (II) no primary bus or device found
[    27.129] 	falling back to /sys/devices/platform/gpu/drm/card0
[    27.129] (II) LoadModule: "glx"
[    27.134] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    27.224] (II) Module glx: vendor="X.Org Foundation"
[    27.225] 	compiled for 1.20.4, module version = 1.0.0
[    27.225] 	ABI class: X.Org Server Extension, version 10.0
[    27.225] (II) LoadModule: "fbturbo"
[    27.228] (II) Loading /usr/lib/xorg/modules/drivers/fbturbo_drv.so
[    27.232] (II) Module fbturbo: vendor="X.Org Foundation"
[    27.232] 	compiled for 1.20.3, module version = 0.5.1
[... 165 lines cut ...]
[    28.906] (II) event2  - vc4: device is a pointer
[    28.906] (II) event2  - vc4: device is a keyboard
[    28.925] (**) vc4: Applying InputClass "evdev keyboard catchall"
[    28.925] (**) vc4: Applying InputClass "libinput keyboard catchall"
[    28.925] (II) Using input driver 'libinput' for 'vc4'
[    28.925] (**) vc4: always reports core events
[    28.925] (**) Option "Device" "/dev/input/event1"
[    28.925] (**) Option "_source" "_driver/libinput"
[    28.925] (II) libinput: vc4: is a virtual subdevice
[    28.925] (**) Option "config_info" "udev:/sys/devices/platform/soc/fef00700.hdmi/rc/rc0/input5/event1"
[    28.926] (II) XINPUT: Adding extended input device "vc4" (type: KEYBOARD, id 10)
[    28.926] (**) Option "xkb_model" "pc105"
[    28.926] (**) Option "xkb_layout" "us"
[    28.926] (**) vc4: Applying InputClass "evdev keyboard catchall"
[    28.926] (**) vc4: Applying InputClass "libinput keyboard catchall"
[    28.926] (II) Using input driver 'libinput' for 'vc4'
[    28.926] (**) vc4: always reports core events
[    28.926] (**) Option "Device" "/dev/input/event2"
[    28.926] (**) Option "_source" "_driver/libinput"
[    28.926] (II) libinput: vc4: is a virtual subdevice
[    28.926] (**) Option "config_info" "udev:/sys/devices/platform/soc/fef05700.hdmi/rc/rc1/input6/event2"
[    28.926] (II) XINPUT: Adding extended input device "vc4" (type: KEYBOARD, id 11)
[    28.926] (**) Option "xkb_model" "pc105"
[    28.926] (**) Option "xkb_layout" "us"
`

alfrix, 2024-05-27:
there was warning for months:
https://github.com/KlipperScreen/KlipperScreen/commit/aa4603af69b7fd59b119d1a9d2ce6f9e8a99367f

technical50, 2024-05-28:
> i see fluiddpi in your login, you probably have python 3.7 test with: `python3 --version` klipperscreen will not work on any distro with python 3.7, due to the need to maintain a working network panel in new versions of the operating system, i needed to switch to python 3.8
> 
> solution1 revert ks and don't update anymore (remove from update manager)
> 
> ```
> cd KlipperScreen
> git reset --hard v0.4.1
> ```
> 
> solution2 update you OS then delete ~/.KlipperScreen-env and reinstall KlipperScreen
> 
> solution3 install a new OS from scratch

Solution 1 I already did I have to keep update language files... so not suitable for me.

Solution 2 I will try today firstly Backup My sd card . after using with "sudo apt full-upgrade" - It look like more easy way.
If success I will update from here.

Solution 3 (Last try after backup all my data)

Thank you for answer.

CoffeeSquirel, 2024-05-28:
I have a similar problem. Fresh install from both Kiauh and of RatOS is not working, when I get my TFT screen working its stuck on console.

In "htop" I can see the klipperscreen restarting over and over again, the PID increasing everytime. My Python version is 3.11.2. 
[KlipperScreen.log](https://github.com/KlipperScreen/KlipperScreen/files/15475038/KlipperScreen.log)
[KlipperScreen.log](https://github.com/KlipperScreen/KlipperScreen/files/15475038/KlipperScreen.log)

I added my KlipperScreen log, let me know if there is anything more I can provide to help anyone who is competent on this.

technical50, 2024-05-29:
I got the system working again.
I briefly list the steps I took.
1- Search for Raspian Os upgrade guide and follow the steps one by one.
2- After Raspian Os Bullseye, delete everything (with kiauh) “Just deleting Klipper Screen is not enough.”
3- After restarting the system, install Klipper-Moonraker-Mainsail respectively and restart it.
4- Finally, install Klipper Screen. (It needs to be opened) If it does not open, check the status with the ```systemctl status KlipperScreen``` command or look at the logs. I experienced the problem of staying on a black screen, and I did this when I restarted after using the ```sudo mv /usr/share/X11/xorg.conf.d/99-fbturbo.conf ~``` command. fixed.
I hope it works for you too.

closed 2024-05-29 by technical50