KlipperScreen/KlipperScreen #1380: [BUG] Hard dependency on NetworkManager means broken RatOS installations.

miklschmidt, 2024-05-27:
### What happened?

Apparently it was decided that all distro's not using NetworkManager should no longer be supported.

### What did you expect to happen instead?

Keep the "old" methods working, or add an option to disable network management entirely.

### How to reproduce this bug?

Install on a distro that doesn't use NetworkManager (such as RatOS).

### Log output

Log file isn't required in this case, this commit dropped support for wpa_supplicant, effectively removing support for Rasperry Pi OS Bullseye: https://github.com/KlipperScreen/KlipperScreen/commit/a80b8810151e6051b8d6dd5b3f2baa3bc54e3a22#diff-b27f73f11b6bbd58c33c636dfc8b29e735974f0c5deab370276d8385879df0fa

miklschmidt, 2024-05-27:
[image]

miklschmidt, 2024-05-27:
This doesn't seem to actually be a thing outside the couple of users that reported this issue, i can also see now that he has the network panel open (it's OK if just that panel breaks). I'm closing this.

closed 2024-05-27 by miklschmidt

Guilouz, 2024-05-29:
@alfrix I have the same issue and after a long time network panel appears, i can see networks but when i try to connect i have "Insufficient privileges" message.

alfrix, 2024-05-29:
> @alfrix I have the same issue and after a long time network panel appears, i can see networks but when i try to connect i have "Insufficient privileges" message.

i'm investigating, the rules worked previously and they should continue working, i have 1 machine that is working correctly and one that shows the privileges issues (the workaround in the docs does work if needed)

Guilouz, 2024-05-29:
@alfrix If that can help, it’s a flsun speeder pad under ubuntu. I have recompiled venv with updated python 3.12, also fixed installation issue with network-manager. Installation is ok but after reboot, klipperscreen take a while to boot and after the issue with privileges