KlipperScreen/KlipperScreen #1381: [BUG] Restarting KlipperScreen removes execute permissions on /dev directory

37wh5r, 2024-05-29:
### What happened?

Whenever I restart KlipperScreen, either via Mainsail service control menu in the web UI or simply be selecting a different icon theme, the /dev directory loses the "x" permission. With the effect that when I ssh into the Raspberry Pi, I am greeted with seven messages "-bash: /dev/null: Permission denied", bash completion does not work, KlipperScreen stops working, and the only way to get to a working system is a "sudo reboot". After the boot, /dev has the 755 permissions again. I find that pretty annoying.

### What did you expect to happen instead?

KlipperScreen (or whichever process is the culprit) should not touch /dev.

### How to reproduce this bug?

already described

### Log output

[KlipperScreen.log](https://github.com/KlipperScreen/KlipperScreen/files/15487766/KlipperScreen.log)

alfrix, 2024-05-29:
i don't see how those things can be related at all, It's more likely to be a system-wide or environment-specific issue

alfrix, 2024-07-14:
Closing due to being stalled and unlikely to be a KS issue

closed 2024-07-14 by alfrix