KlipperScreen/KlipperScreen #1384: [BUG] <title>

NikolayMurha, 2024-05-30:
### What happened?

/home/pi/KlipperScreen/scripts/KlipperScreen-install.sh: line 154: syntax error near unexpected token `}'

### What did you expect to happen instead?

Installation

### How to reproduce this bug?

Try to install using KlipperScreen-install.sh

### Log output

```
/home/pi/KlipperScreen/scripts/KlipperScreen-install.sh: line 154: syntax error near unexpected token `}'
/home/pi/KlipperScreen/scripts/KlipperScreen-install.sh: line 154: `}'

#=======================================================#
 KlipperScreen installation failed!
#=======================================================#
```

HaniAlotaibi, 2024-05-30:
i just had the same issue exactly. my screen was not working (BTT HDMI5), removed the installed version, and when i tried to reinstall it, it  didnt work, direct install via KlipperScreen-install.sh and via Kiauh

dr-pavel, 2024-05-30:
I think we need to add the **fi** operator after 
these lines:
```
 if [ -d $KSENV ]; then
        echo_text "Removing old virtual environment"
        rm -rf ${KSENV}
```
You should get something like this:
```
if [ -d $KSENV ]; then
    echo_text "Removing old virtual environment"
    rm -rf ${KSENV}
    fi
```
then run the script:
`KlipperScreen-install.sh`

In my case, the installation was successful.

teunvandenburg, 2024-05-30:
> I think we need to add the **fi** operator after these lines:
>
> ```
>  if [ -d $KSENV ]; then
>         echo_text "Removing old virtual environment"
>         rm -rf ${KSENV}
> ```
>
> You should get something [...]

Works for me aswel thanks a lot.

Safiro1, 2024-05-30:
I having the exact same issue when trying manual installation or using KIAUH.
Please let us know when this is fixed.
Thanks!

Safiro1, 2024-05-30:
> Works for me aswel thanks a lot.

Where did you add those conditionals?

Smurfj3, 2024-05-31:
Same problem here, fixing the script made the script go, however half way the installation I now get the error "KlipperScreen-install.sh: line 124: /home/android/.KlipperScreen-env/bin/activate: No such file or directory"

Smurfj3, 2024-05-31:
This breaks the Kiauh installation helper completely because even changing the KlipperScreen-Install.sh does not make a difference because Kiauh will redownload the files from github. Hopefully this gets updated soon :D

arraylabs, 2024-05-31:
Stuck in Kiauh with this as well

Smurfj3, 2024-05-31:
> Stuck in Kiauh with this as well

Yeah I got it working but it was a bit of a B

brunobold, 2024-05-31:
#1385 Pull request submitted with fix.

alfrix, 2024-05-31:
i was busy and did not have the time to even look at the repo until now, thanks bruno, and sorry to everyone for the inconvenience.

closed 2024-05-31 by alfrix