KlipperScreen/KlipperScreen #1386: [BUG] <Klipperscreen Wifi Manger not work>

technical50, 2024-06-01:
### What happened?

When More manu and click wifi it appears error in menu "failed to initialize sdbus, failed to start network manager"
[image]

### What did you expect to happen instead?

Wifi configration change.. - Ip adress check etc.

### How to reproduce this bug?

Screen stuck on error and not able to back.

### Additional information:

Latest Debian Latest Mainsail .

### Log output

2024-06-01 16:35:40,011 [printer.py:configure_cameras()] - Cameras: []
2024-06-01 16:35:40,036 [screen.py:init_klipper()] - Moonraker info {'klippy_connected': True, 'klippy_state': 'ready', 'components': ['secrets', 'template', 'klippy_connection', 'jsonrpc', 'internal_transport', 'application', 'websockets', 'dbus_manager', 'database', 'file_manager', 'authorization', 'klippy_apis', 'shell_command', 'machine', 'data_store', 'proc_stats', 'job_state', 'job_queue', 'history', 'http_client', 'announcements', 'webcam', 'extensions', 'octoprint_compat', 'update_manager'], 'failed_components': [], 'registered_directories': ['config', 'logs', 'gcodes', 'config_examples', 'docs'], 'warnings': [], 'websocket_count': 1, 'moonraker_version': 'v0.8.0-377-g4fe89d5', 'missing_klippy_requirements': [], 'api_version': [1, 4, 0], 'api_version_string': '1.4.0'}
2024-06-01 16:35:40,115 [printer.py:log_counts()] - Klipper version: v0.12.0-207-g6cd17420
2024-06-01 16:35:40,116 [printer.py:log_counts()] - # Extruders: 1
2024-06-01 16:35:40,116 [printer.py:log_counts()] - # Temperature devices: 2
2024-06-01 16:35:40,117 [printer.py:log_counts()] - # Fans: 2
2024-06-01 16:35:40,117 [printer.py:log_counts()] - # Output pins: 2
2024-06-01 16:35:40,117 [printer.py:log_counts()] - # PWM tools: 0
2024-06-01 16:35:40,117 [printer.py:log_counts()] - # Leds: 0
2024-06-01 16:35:40,376 [KlippyWebsocket.py:object_subscription()] - Sending printer.objects.subscribe
2024-06-01 16:35:40,380 [files.py:set_gcodes_path()] - Gcodes path: /home/pi/printer_data/gcodes
2024-06-01 16:35:40,381 [screen.py:init_klipper()] - Printer initialized
2024-06-01 16:35:40,382 [printer.py:change_state()] - Changing state from 'disconnected' to 'ready'
2024-06-01 16:35:40,382 [printer.py:change_state()] - Adding callback for state: ready
2024-06-01 16:35:40,406 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 900
2024-06-01 16:35:40,430 [screen.py:set_screenblanking_timeout()] - Using DPMS
2024-06-01 16:35:40,444 [KlippyWebsocket.py:get_file_list()] - Sending server.files.list
2024-06-01 16:35:40,447 [screen.py:_remove_all_panels()] - Removing all panels
2024-06-01 16:35:40,455 [screen.py:_load_panel()] - Loading panel: main_menu
2024-06-01 16:35:40,632 [main_menu.py:__init__()] - ### Making MainMenu
2024-06-01 16:35:40,696 [main_menu.py:add_device()] - Adding device: extruder
2024-06-01 16:35:40,711 [main_menu.py:add_device()] - Adding device: heater_bed
2024-06-01 16:35:40,727 [main_menu.py:add_device()] - Adding device: temperature_sensor raspberry_pi
2024-06-01 16:35:40,787 [screen.py:attach_panel()] - Current panel hierarchy: main_menu
2024-06-01 16:35:40,835 [printer.py:init_temp_store()] - Temp store: ['heater_bed', 'temperature_sensor raspberry_pi', 'extruder']
2024-06-01 16:35:40,859 [screen.py:init_tempstore()] - Temperature store size: 1200
2024-06-01 16:35:43,591 [screen.py:_go_to_submenu()] - #### Go to submenu more
2024-06-01 16:35:43,592 [screen.py:_go_to_submenu()] - #### Menu __main
2024-06-01 16:35:43,601 [screen.py:_load_panel()] - Loading panel: menu
2024-06-01 16:35:43,686 [menu.py:evaluate_enable()] - moonraker connected True
2024-06-01 16:35:44,043 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > more
2024-06-01 16:35:44,058 [menu.py:arrangeMenuItems()] - X > more bedmesh
2024-06-01 16:35:44,069 [menu.py:arrangeMenuItems()] - X > more retraction
2024-06-01 16:35:44,080 [menu.py:arrangeMenuItems()] - X > more led
2024-06-01 16:35:44,101 [menu.py:arrangeMenuItems()] - X > more power
2024-06-01 16:35:44,107 [menu.py:arrangeMenuItems()] - X > more camera
2024-06-01 16:35:44,136 [menu.py:arrangeMenuItems()] - X > more spoolman
2024-06-01 16:35:44,136 [menu.py:evaluate_enable()] - moonraker connected True
2024-06-01 16:35:45,809 [screen.py:_load_panel()] - Loading panel: network
2024-06-01 16:35:46,626 [sdbus_nm.py:ensure_nm_running()] - Starting NetworkManager service...
2024-06-01 16:35:46,771 [network.py:__init__()] - Failed to initialize
Traceback (most recent call last):
  File "/home/pi/KlipperScreen/panels/network.py", line 21, in __init__
    self.sdbus_nm = SdbusNm(self.popup_callback)
  File "/home/pi/KlipperScreen/ks_includes/sdbus_nm.py", line 93, in __init__
    self.ensure_nm_running()
  File "/home/pi/KlipperScreen/ks_includes/sdbus_nm.py", line 122, in ensure_nm_running
    raise RuntimeError("Failed to start NetworkManager service")
RuntimeError: Failed to start NetworkManager service
2024-06-01 16:35:46,811 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > more > network
2024-06-01 23:33:42,650 [screen.py:_menu_go_back()] - #### Menu go back
[... 82 lines cut ...]
    self._screen._menu_go_back()

  File "/home/pi/KlipperScreen/screen.py", line 531, in _menu_go_back
    self._remove_current_panel()

  File "/home/pi/KlipperScreen/screen.py", line 524, in _remove_current_panel
    self.panels[self._cur_panels[-1]].deactivate()

  File "/home/pi/KlipperScreen/panels/network.py", line 360, in deactivate
    if self.sdbus_nm.wifi:
NoneType: None
2024-06-01 23:33:46,977 [screen.py:_menu_go_back()] - #### Menu go back
2024-06-01 23:33:46,979 [functions.py:logging_exception_handler()] - Uncaught exception <class 'AttributeError'>: 'NoneType' object has no attribute 'wifi'
  File "/home/pi/KlipperScreen/panels/base_panel.py", line 208, in back
    self._screen._menu_go_back()

  File "/home/pi/KlipperScreen/screen.py", line 531, in _menu_go_back
    self._remove_current_panel()

  File "/home/pi/KlipperScreen/screen.py", line 524, in _remove_current_panel
    self.panels[self._cur_panels[-1]].deactivate()

  File "/home/pi/KlipperScreen/panels/network.py", line 360, in deactivate
    if self.sdbus_nm.wifi:
NoneType: None

EyalRotem, 2024-06-02:
i am having this issue too. the PI is disonnected from the network, mid print, when i go to klipperscreen network menu, i get stuck there with the same message

alfrix, 2024-06-02:
The network panel needs network-manager to work, it's included in the raspberryOS, if it's not in your OS you need to install it for the panel to work, the installer now will ask you, i've been busy and couldn't finish the supplementary changes for those using other distros, i've just pushed a set of patches that should be enough for most people

technical50, 2024-06-02:
> The network panel needs network-manager to work, it's included in the raspberryOS, if it's not in your OS you need to install it for the panel to work, the installer now will ask you, i've been busy [...]

Still have issues already installed network-manager.
[image]
[image]

mdurfee, 2024-06-02:
Having the same issue. Prior to the newest release it said "failed to init sd bus" as in the first post here. Network Manager is installed.

daniel78ktn, 2024-06-03:
This work for me:
https://forum-raspberrypi.de/forum/thread/57368-networkmanager-auf-raspian11-debian11-bullseye-am-terminal-einrichten-serial-ter/

I did what the code said in the 3rd window:  from 1 sudo apt-get update to step 6
now i have the network Window

alfrix, 2024-06-03:
i've made some changes to the installer that should work better, based on what the raspi-config script does, using raspi-config to enable network-manager is valid for those who want to use it

technical50, 2024-06-03:
> i've made some changes to the installer that should work better, based on what the raspi-config script does, using raspi-config to enable network-manager is valid for those who want to use it

Yes I was tryed Firstly raspi-config side correction after my wifi disconnect and not able to connect. After I did re-install using by kiauh (klipperscreen) during installition it asks network manager and I was select but still wifi not work.
[image]
Wifi manager work but pirivilage issue has and not solve Clicking "Accept"
I was connect again via Lan Cable and I used this command (sudo nmcli dev wifi connect network-ssid password "network-password) It connected but Im not able change Network and manage still pirivilage issues have...

Machine73, 2024-06-04:
I've been having the same issues. I am now on my 6th clean install of klipper on my Voron 2.4. PI no longer can connect to network with a full "default" install of KS. As long as I don't install network manager, my PI and machine connect, and work as it did before. Just no network manager. If I hit the network button, my PW shows "insufficient privileges" them locks up the screen, having to do a power cycle.
It has come down to just doing a full install minus network manager to get things running again.

vakeiros, 2024-06-06:
I've been with same issues as all previous, first I did a clean install on a new machine and when I installed klipperscreen (with network-manager) my Pi couldn't connect to WiFi, if I go to Network menu if sees all the networks and if I try to connect to one of them it says "Insuficient privilledges detected". then if I do the "sudo nmcli dev wifi connect network-ssid password "network-password" with a ethernet cable connected it will connect to the network I use in the command but then if I go to the Network menu in the screen no wifi networks are seen, not even the one I'm connected to.

technical50, 2024-06-07:
After Re-install using "kiauh" Permission problems was fixed - Updating not fix permissions !

closed 2024-06-07 by technical50

DrumClock, 2024-07-20:
Hi, even if I have a static wlan address entered in the Rpi, klippersreen manager overwrites it with the one assigned by the DHCP server of the wifi router.

HSS:  `sudo nano /etc/dhcpcd.conf`

```
interface wlan0
static ip_address=192.168.0.32/24
static routers=192.168.0.1
static domain_name_servers=192.168.0.1
```

Hi, even if I have a static wlan address entered in the Rpi, klippersreen manager overwrites it with the one assigned by the DHCP server of the wifi router.

in Rpi:

sudo nano /etc/dhcpcd.conf

interface wlan0
static ip_address=192.168.0.32/24
static routers=192.168.0.1
static domain_name_servers=192.168.0.1

**Where do I set a wlan static IP address for klippersreen?**

tahsin3323, 2024-08-24:
> This work for me: https://forum-raspberrypi.de/forum/thread/57368-networkmanager-auf-raspian11-debian11-bullseye-am-terminal-einrichten-serial-ter/
> 
> I did what the code said in the 3rd window: from 1 sudo apt-get update to step 6 now i have the network Window

Yes this Helps. i had to need  extra permission,

For activating network manager :

Connect pi via ethernet cable
Open Putty and login

`sudo apt-get update
sudo apt-get upgrade -y
sudo apt-get install network-manager
sudo raspi-config
6 Advanced Options -> AA Network Config -> NetworkManager -> Ok 
sudo reboot
`
For Autheration :  [source](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/Network/#permission-error)

`mkdir -p /etc/NetworkManager/conf.d
sudo nano /etc/NetworkManager/conf.d/any-user.conf`

in the editor paste this:

`[main]
auth-polkit=false`

Then restart the service (or reboot):
`systemctl restart NetworkManager.service
systemctl restart KlipperScreen.service`

Lenorde, 2024-12-07:
> For activating network manager :
> 
> Connect to RPi via ethernet cable Open Putty and login
> 
> * `sudo apt-get update
> * sudo apt-get upgrade -y
> * sudo apt-get install network-manager
> * sudo raspi-config
> * 6 Advanced Options -> AA Network Config -> NetworkManager -> Ok
> * sudo reboot
> 
> `
> 
> For Autheration : [source](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/Network/#permission-error)
> 
> * `mkdir -p /etc/NetworkManager/conf.d
> * sudo nano /etc/NetworkManager/conf.d/any-user.conf`
> 
> in the editor paste this:
> 
> `[main] auth-polkit=false`
> 
> Then restart the service (or reboot): `systemctl restart NetworkManager.service systemctl restart KlipperScreen.service`

Hi, I didn't have access to the network manager in Klipperscreen either.
After doing the first part, it displayed the available Wi-Fi but didn't had the authorization to connect. I switched to an ethernet connection to do the last two commands but now the Network Manager is off again, and I don't have access to the printer with ethernet.
How can I resolve that without connection?