KlipperScreen/KlipperScreen #1387: [FR] Make WIFI network optional

DirkErdmann, 2024-06-02:
### Requested feature

As a user who is using LAN via cable in general I want to disable the occurrence of the WIFI network section.

I see two advantages:

1. if this is an opt-in via settings menu or via `Klipperscreen.conf` the user can decide and e.g. will have a cleaner config screen
2. if a user is using a Linux environment without Network manager (especially because it's already optional for Klipperscreen) it will avoid problems for existing network configurations (LAN and/or WIFI). If the user enables the function via settings dialogue, Klipperscreen can post-install the required packages.

### Is your feature request related to a problem?

- see advantage 2
- when removing Network Manager the old network config will not be usable anymore (e.g. eth0 config will be deleted)

### Additional information

Would be great to have it at least as opt-in wo decide if it's available on the screen or not.

miklschmidt, 2024-06-02:
Second that! RatOS has a whole setup around wifi/hotspotting and doesn't use network manager (on purpose), so would be nice to disable this by default for us.

alfrix, 2024-06-03:
the network panel shows ip and host info if no wifi is found

i want the user to be informed, so i will not hide it by default for now at least until most buster users have upgraded or understood

you should be able to override any menu entry from KlipperScreen.conf for example:
```
[menu __main more network]
enable: False
```

miklschmidt, 2024-06-03:
> i want the user to be informed, so i will not hide it by default for now at least until most buster users have upgraded or understood

Yeah i don't think it should be hidden by default either, just wanted a possibility to opt-out. Network Manager isn't default on Bullseye btw, that didn't happen before Bookworm. There are lots of people not on bookworm yet, and there are other distro's (particular on other SBCs) than Debian/Raspberry Pi OS. I miss the support for the other network tools, but i guess i can live with that if i can disable the Network panel.

>  you should be able to override any menu entry from KlipperScreen.conf for example:
Perfect, that's actually all i needed. Should've read the docs closer 🤦‍♂️

Erde-E, 2024-06-03:
> the network panel shows ip and host info if no wifi is found
> 
> i want the user to be informed, so i will not hide it by default for now at least until most buster users have upgraded or understood
> 
> you should be able to override any menu entry from KlipperScreen.conf for example:
> 
> ```
> [menu __main more network]
> enable: False
> 
> [menu __print network]
> enable: False
> 
> [menu __splashscreen network]
> enable: False
> ```
> 
> edit: added all 3 states in which the panel appear just in case you don't know how the menu works

Thanks for your response - atm it’s ok for me as well.

Maybe you can be so kind and tell when the Networkmanager is uses - at the moment the network panel is accessed? (sorry for not diggin into the code)
I’m thinking about one additional step as a workaround or quick solution in case of differing network environments:
-> disabling the NM service right after initial installation and hide the menu from the UI

Thanks a lot!

alfrix, 2024-06-03:
networkmanager will only be interfaced when using the panel
since networkmanager is already optional in the installer this is considered as complete

closed 2024-06-03 by alfrix

miklschmidt, 2024-06-03:
@alfrix when you said it was optional i went and looked at the script again because it’s always installed for me when installing unattended, then i saw the problem. “$NETOWRK” 😅

alfrix, 2024-06-03:
@miklschmidt oops, i'll fix the typo