KlipperScreen/KlipperScreen #1395: [BUG] printer button malfunction

Schulli73, 2024-06-07:
### What happened?

I wanted to select a new print file using the Klipper screen. I pressed the print button, then the screen crashed and a new one started.

### What did you expect to happen instead?

That the selection menu for the print files appears.

### How to reproduce this bug?

Push the printer button.

### Additional information:

[image]

### Log output

[KlipperScreen.log](https://github.com/user-attachments/files/15744485/KlipperScreen.log)

alfrix, 2024-06-07:
not a bug, the print panel was renamed to gcodes
you placed the menu entries in the config is your responsability to maintain them
the recommendation is to override things that you want to change, but not to place all the entries like you did here.

closed 2024-06-07 by alfrix

Schulli73, 2024-06-10:
Ich habe keine Änderung dran gemacht. Ich habe das Problem zeitgleich an zwei Druckern bekommen. Wie kann ich das Problem reparieren? Ich habe keine Ahnung was ich machen muss. 

I haven't made any changes. I got the problem on two printers at the same time. How can I fix the problem? I have no idea what to do.

tigerwoodsDE, 2024-07-31:
@alfrix  where are the entries located? Which entries need to be renamed from the print panel to gcodes? I'm a little confused and can't find it...

kyzorr, 2024-11-07:
@tigerwoodsDE 
go to yout KlipperScreen.cfg and search "[menu __main print]" and change the "panel: print" to "panel: gcodes"

```
[menu __main print]
name: {{ gettext('Print') }}
icon: folder
panel: gcodes
```

and if your icon is still set to "print" change it to one of those: printe, sd, folder, files, files or another file from `~/KlipperScreen/styles/material-dark/images`
the panel-name are stored here `~/KlipperScreen/panels/`

tigerwoodsDE, 2024-11-07:
@kyzorr 
nice! It works again, thanks for the support!