KlipperScreen/KlipperScreen #1400: Python requirement...

3Dgraf64, 2024-06-12:
### What happened?

After doing the update the system tells me that my version of Python (3.7) is out of date.  Whatever I do to fix it (3.8, 3.12), the same error persists.  Even if there is no more Python (purge).  I will have to reinstall my entire system.  It’s annoying.  There is something missing to support the user.

### What did you expect to happen instead?

Something to refer to resolve the issue properly.

### How to reproduce this bug?

Just doing the update while current python version is 3.7.

### Log output

[KlipperScreen.log](https://github.com/user-attachments/files/15807656/KlipperScreen.log)

technical50, 2024-06-17:
It was explained before please check closed issues first !
https://github.com/KlipperScreen/KlipperScreen/issues/1379

3Dgraf64, 2024-06-17:
So in short:  I have to redo everything to keep all updated.  Thanks.

closed 2024-06-17 by 3Dgraf64