KlipperScreen/KlipperScreen #1403: [BUG] Z-Calibrate starts BED_MESH_CALIBRATE

hapklaar, 2024-06-13:
### What happened?

Wanted to run a Z-Calibrate from Klipperscreen, but hitting Start starts a BED_MESH_CALIBRATE instead. After this all buttons on the Z Calibrate screen became unavailable.

### What did you expect to happen instead?

Expected the head to go to the center and 5 mm above the bed, ready to manually do a Z calibration

### How to reproduce this bug?

Start Z-Calibrate from Klipperscreen menu

### Additional information:

Ratrig 3.0 400 with Beacon eddy current probe

KlipperScreen v0.4.2-11-g629c86d9

### Log output

```
2024-06-13 19:27:17,313 [printer.py:get_probe()] - Probe type: probe
2024-06-13 19:27:17,313 [zcalibrate.py:__init__()] - Offset X:0.0 Y:0.0 Z:1.3
2024-06-13 19:27:17,336 [zcalibrate.py:set_functions()] - Mesh: {'speed': '300', 'horizontal_move_z': '5', 'mesh_min': '20,30', 'mesh_max': '365,360', 'probe_count': '14, 14', 'fade_start': '1.0', 'fade_end': '10.0', 'mesh_pps': '2,2', 'algorithm': 'bicubic', 'bicubic_tension': '.2'}
2024-06-13 19:27:17,337 [zcalibrate.py:set_functions()] - Available functions for calibration: ['mesh']
2024-06-13 19:27:17,361 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > more > zcalibrate
2024-06-13 19:27:18,889 [printer.py:init_temp_store()] - Temp store: ['temperature_sensor Octopus', 'temperature_sensor raspberry_pi', 'heater_bed', 'temperature_sensor beacon_coil', 'extruder']
2024-06-13 19:27:23,907 [printer.py:init_temp_store()] - Temp store: ['temperature_sensor Octopus', 'temperature_sensor raspberry_pi', 'heater_bed', 'temperature_sensor beacon_coil', 'extruder']
2024-06-13 19:27:24,541 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: SET_GCODE_OFFSET Z=0
2024-06-13 19:27:24,543 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: BED_MESH_CALIBRATE
```

alfrix, 2024-06-21:
not a bug, unsupported probe type, the behavior could be improved, i wanted to throw an easier path to manually calibrate if you did not have a probe, but it will likely be imposible to maintain current behaviour with devs of modules adding custom names to whats essentially just a probe_calibrate

closed 2024-06-21 by alfrix

hapklaar, 2024-06-21:
I do very much like it that when I start a `BEACON_CALIBRATE` from cmdline, the correct manipulation interface shows on KS, but doing a `BED_MESH_CALIBRATE` upon selecting a Z-Calibrate rather unexpected. Maybe we can swap menu items when a Beacon is detected, or make it configurable in config?

alfrix, 2024-06-21:
> but doing a BED_MESH_CALIBRATE upon selecting a Z-Calibrate rather unexpected

i've added the method that will be used to the button name.

if you want to start it using custom commands do something like:

```
[gcode_macro PROBE_CALIBRATE]
rename_existing: PROBE_CALIBRATE_BASE
gcode:
    BEACON_CALIBRATE
```

i will not be adding custom module commands to mainline, i will probably create some sort of custom config at a later point

commit 468901d9 (2024-06-21): bed_mesh and zcalibrate remove easy manual transition when no probe due to raising use unknwown probe commands, and always show what is the default if there is only 1 command in zcal close #1403