KlipperScreen/KlipperScreen #1405: [FR] Filament type shown in file preview! and maybe static menus instead of scrolling?

cptcl, 2024-06-14:
### Requested feature

Hello
I would like to see the filament type (like PLA, PETG, ASA, ..) in the fileviewer / preview of the gcode files.

And a question, but this is more specific.
Possible to have Klipperscreen, instead of smoth scrolling, controlled by an up, down and back button? 

i.e. u open a folder and can see the first 6 files in a small preview, on pressing the small preview u would see all details of the file.
For the next 6 files u just use an slim, but longer up and down button on the right side, and go back with a button between (or above) the up and down button. (Also for menu)
This maybe would also save some resources.

### Is your feature request related to a problem?

Not a real problem but 2 little one. I have a Qidi X-Max 3 v2.
The software (OS, Klipper [is using 0.10], Moonraker and Fluidd) is very out of date, and Qidi is not providing an official update.
In order to get those to newest version i need to replace the display, cause Qidi made some changes to Klipper, Moonraker and Fluidd to get their homebrew display running (even tho i rly like the display and how it shows everything). 
Have buyed a waveshare 5" touch display, using an pi zero 2 w, but this one (or two) feature is holding me back to convert it already.
I really like how my x-max 3 is displaying the folders & gcode files as i said. Will provide a screenshot and a drawing within next days, about what i mean.
Also while scolling, sometimes it recognizes a fast scroll for a klick. so i have to go back once, and need to scroll again.

### Describe alternatives you have considered

Putting the filament type inside the name of the file, but sometimes the filename is than really long. 
Also if u forgot to do so and have to print it again later.. u maybe dont remember it.

alfrix, 2024-06-21:
> I would like to see the filament type (like PLA, PETG, ASA, ..) in the fileviewer / preview of the gcode files.

it's already possible in the list mode

> Possible to have Klipperscreen, instead of smoth scrolling, controlled by an up, down and back button?

there are optional arrows for the scrollbar in the settings, smooth scrolling will always be there, the arrow scroll-step may be tweaked if i discover any workaround but all attempts have failed at the moment, but paginating everything is not in my goals.

> on pressing the small preview u would see all details of the file.

i have sort of done that but i'm testing in different orientations and deciding on the details

cptcl, 2024-06-22:
Ah.. i updated rn and its there. thank u so much.
Could we have the filament type also in the big view? when u klick on the image u have the preview and the print time.. possible to have the print details there too? like.. putting the preview a bit to the sid e and have all details i.e. on the right side next to the preview.

alfrix, 2024-06-22:
> Ah.. i updated rn and its there. thank u so much. Could we have the filament type also in the big view? when u klick on the image u have the preview and the print time.. possible to have the print [...]

i've made that a few weeks a ago and forgot to test it, this reminded me of it, i will merge it after testing

[image]

cptcl, 2024-06-22:
Wow... this is (would be) perfect. u are awsome!

alfrix, 2024-07-14:
closing since c9f4f3b06654f70f7bef6ee5f55635648795e734

closed 2024-07-14 by alfrix