KlipperScreen/KlipperScreen #1407: [BUG] <title>

Levitron86, 2024-06-19:
### What happened?

Hi dear Klipper Fans,

after todays update to v0.4.2-12-g98c6ca5d

I get an Error when pushing the print button in KlipperScreen -> print.py Panel not found

Obviously the panel is no more in the panels folder on my pi and @git aswell. Please check if this is ok, or how i could get to the previous version, where everything works fine for me. THX

2024-06-19 21:31:12,985 [screen.py:_load_panel()] - Loading panel: print
2024-06-19 21:31:12,986 [screen.py:_load_panel()] - Panel print does not exist
2024-06-19 21:31:12,990 [screen.py:show_error_modal()] - Showing error modal: Unable to load panel print [Errno No such file or directory] 
/home/pi/KlipperScreen/panels/print.py

Traceback (most recent call last):
  File "/home/pi/KlipperScreen/screen.py", line 323, in show_panel
    self.panels[panel_name] = self._load_panel(panel).Panel(self, title, **kwargs)
  File "/home/pi/KlipperScreen/screen.py", line 304, in _load_panel
    raise FileNotFoundError(os.strerror(2), "\n" + panel_path)
FileNotFoundError: [Errno No such file or directory] 
/home/pi/KlipperScreen/panels/print.py

### What did you expect to happen instead?

The printer should start the print job

### How to reproduce this bug?

All time i press print on the screen the error pop up.

### Log output

[KlipperScreen.log](https://github.com/user-attachments/files/15905352/KlipperScreen.log)

Guilouz, 2024-06-20:
https://github.com/KlipperScreen/KlipperScreen/issues/1395

alfrix, 2024-06-21:
thank you Guilouz

closed 2024-06-21 by alfrix