KlipperScreen/KlipperScreen #1410: [BUG] BTT Pi TFT flashes white on screen wake up after an update

Bimbovadasz, 2024-06-22:
### What happened?

Hi!

After an update my BTT Pi TFT50 V2.1 flashes white when I touch the screen to wake it up. Unfortunatelly this update was some time ago, about 2 month maybe, I just haven't got the time to write this report until now. Since then I regurarly updated Klipperscreen, but the problem is still present. 

[image]

If I disable the screen DPMS option the white flash disappears, but then the screen doesn't turn off, just remains black.
The screen is capadble of turning off, previously it worked perfectly.

Thank you for the help!

### What did you expect to happen instead?

The screen to wake up without a white flash when touched. 

### How to reproduce this bug?

Touch the screen after it went to sleep. 

### Log output

[klipperscreen_log.zip](https://github.com/user-attachments/files/15937299/klipperscreen_log.zip)

alfrix, 2024-06-22:
screen wake from sleep is controlled by the video drivers, i don't think there is anything that can be done from my side

alfrix, 2024-07-14:
Closing since this is not something fixable by ks it's a driver thing when the screen wakes up

closed 2024-07-14 by alfrix