KlipperScreen/KlipperScreen #1413: I am unable to install KlipperScreen

DJKyomi, 2024-06-25:
### What happened?

The installation of Klipperscreen always fails with this error
[Klipperscreen_error.txt](https://github.com/user-attachments/files/15973879/Klipperscreen_error.txt)

### What did you expect to happen instead?

i can to nothing 

### How to reproduce this bug?

?

### Additional information:

Orange PI PC + , Armbian latest Release 

### Log output

[logs-20240625-174111.zip](https://github.com/user-attachments/files/15973903/logs-20240625-174111.zip)

alfrix, 2024-06-25:
you may need to install libsystemd 

apt install libsystemd0

DJKyomi, 2024-06-26:
is the latest version 
sudo apt install systemd 
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
systemd ist schon die neueste Version (249.11-0ubuntu3.12).
systemd wurde als manuell installiert festgelegt.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.

sudo apt install libsystemd0
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
libsystemd0 ist schon die neueste Version (249.11-0ubuntu3.12).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.

alfrix, 2024-06-26:
in their docs says it's not required since 0.8
try also installing the headers
apt install libsystemd-dev

DJKyomi, 2024-06-26:
So the error is only under Armbian Ubuntu Jammy, I have now deleted the entire Orangepi and installed Armbian 
Debian Bookworm and it works wonderfully

alfrix, 2024-07-14:
close with [cee185e](https://github.com/KlipperScreen/KlipperScreen/commit/cee185e21e6cf7fc2039d3ace8591a8b81ff1ae7)

closed 2024-07-14 by alfrix