KlipperScreen/KlipperScreen #1416: [FR] Option to adjust temperature when loading & unloading filament

steveharman, 2024-06-28:
### Requested feature

**OVERVIEW**
KlipperScreen has options for loading & unloading filament on the Extrude screen.  There is a minimum temperature requirement before either of these functions will operate.

**ISSUE**
When loading / unloading filament I am presented with an alert if the extruder temperature is not high enough.  I then need to exit the Extrude screen, visit the temperature screen, set the extruder to a reasonable temperature (190deg+?), then return to the Extrude screen and start my load / unload again.

**REQUEST**
IF extruder has not reached the minimum temperature required for loading / unloading filament THEN present the user with a popup dialog "Set minimum temperature? [YES]   [NO]" when they attempt to load / unload filament.

This will set the extruder to the minimum required temperature without any further input from the user, who can resume their load / unload when the temperature is reached, without having to leave & return to the Extrude screen.

_Optional_:  Perhaps the loading / unloading could begin automatically once the extruder has reached temperature - meaning even fewer taps for the user to perform. Case:

- User:  <load filament>
- KlipperScreen: "Set minimum temperature? [YES]   [NO]"
- User: <yes>
- KlipperScreen: <waits for temperature to be reached, begins the loading process>.

### Is your feature request related to a problem?

No.

### Describe alternatives you have considered

Manually existing the screen, setting the temperature, then returning to the Extrude screen.

alfrix, 2024-07-06:
thank you, i really liked your idea, i implemented it slightly differently,
if the temp was set above minimum but not reached it will insert a temp wait
if the temp was not set or set too low, it will just take you to the temp panel with a message so you can set it as you want and then go back if you press extrude it should wait for your temp before extruding

closed 2024-07-06 by alfrix

adelyser, 2024-07-07:
> thank you, i really liked your idea, i implemented it slightly differently, if the temp was set above minimum but not reached it will insert a temp wait if the temp was not set or set too low, it [...]

This change broke both of my KlipperScreen installs.  It appears that `min_extrude_temp` isn't pulled in from the main printer config. I specifically added it as I was using the default 170 value,  and it still wasn't pulled in. When I try to run any extrude commands, with temp set at 250C, the python errors appear.

FYI, I reverted to the commit before this and things are working fine again.

```
2024-07-06 14:15:04,653 [screen.py:_load_panel()] - Loading panel: extrude
2024-07-06 14:15:04,747 [screen.py:attach_panel()] - Current panel hierarchy: main_menu > extrude
2024-07-06 14:15:04,749 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: SET_FILAMENT_SENSOR SENSOR=runout_sensor ENABLE=1
2024-07-06 14:15:04,750 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: SET_FILAMENT_SENSOR SENSOR=encoder_sensor ENABLE=1
2024-07-06 14:15:08,115 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'min_extrude_temp'
  File "/home/pi/KlipperScreen/panels/extrude.py", line 276, in check_min_temp
    min_extrude_temp = float(self._printer.config[self.current_extruder]['min_extrude_temp'])
                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
NoneType: None
2024-07-06 14:15:10,366 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'min_extrude_temp'
  File "/home/pi/KlipperScreen/panels/extrude.py", line 276, in check_min_temp
    min_extrude_temp = float(self._printer.config[self.current_extruder]['min_extrude_temp'])
                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
NoneType: None
2024-07-06 14:15:50,415 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'min_extrude_temp'
  File "/home/pi/KlipperScreen/panels/extrude.py", line 276, in check_min_temp
    min_extrude_temp = float(self._printer.config[self.current_extruder]['min_extrude_temp'])
                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
NoneType: None
2024-07-06 14:17:10,669 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'min_extrude_temp'
  File "/home/pi/KlipperScreen/panels/extrude.py", line 276, in check_min_temp
    min_extrude_temp = float(self._printer.config[self.current_extruder]['min_extrude_temp'])
                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
NoneType: None
2024-07-06 14:17:25,759 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'min_extrude_temp'
  File "/home/pi/KlipperScreen/panels/extrude.py", line 276, in check_min_temp
    min_extrude_temp = float(self._printer.config[self.current_extruder]['min_extrude_temp'])
                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
NoneType: None
```

commit 7bf4e5ba (2024-07-06): feat(extrude): wait for temp or open temp panel automatically close #1416