KlipperScreen/KlipperScreen #1419: [BUG] Unable to connect to Wi-Fi after installing KlipperScreen

trebory6, 2024-07-01:
### What happened?

After installing Klipperscreen, I am unable to connect to the wifi at all.

Previously to installing Klipperscreen wifi worked fine.

Considering it worked fine before and now it does not leads me to think there was an issue with the Klipperscreen installation.

### What did you expect to happen instead?

The raspberry pi to connect to the wifi.

### How to reproduce this bug?

Plug the pi in and try to connect with WiFi.

### Additional information:

So if I connect the pi to Ethernet and ssh into the pi and do the `sudo raspi-config` command and enter the SSID and password there, I get the following error:

`There was an error running option S1 Wireless LAN`

So something seems to have messed with my NetworkManager.

Also to note, my entire log file couldn't fit in the log section. If previous log is needed I can try to upload a different way.

### Log output

2024-06-30 18:54:34,656 [functions.py:log_start()] - --------------------KlipperScreen Log Start--------------------
2024-06-30 18:54:34,657 [functions.py:log_start()] - KlipperScreen Version: v0.4.3-1-g03a3837
2024-06-30 18:54:34,657 [functions.py:log_start()] - Python version: 3.9
2024-06-30 18:54:34,771 [config.py:get_config_file_location()] - Passed config (-c): .
2024-06-30 18:54:34,771 [config.py:__init__()] - Config path location: /home/Majestic4329/printer_data/config/KlipperScreen.conf
2024-06-30 18:54:34,773 [config.py:_include_config()] - Parsing files: ['/home/Majestic4329/KlipperScreen/config/main_menu.conf']
2024-06-30 18:54:34,782 [config.py:_include_config()] - Parsing files: ['/home/Majestic4329/KlipperScreen/config/splash_menu.conf']
2024-06-30 18:54:34,785 [config.py:_include_config()] - Parsing files: ['/home/Majestic4329/KlipperScreen/config/print_menu.conf']
2024-06-30 18:54:34,790 [config.py:_include_config()] - Parsing files: ['/home/Majestic4329/KlipperScreen/config/move_menu.conf']
2024-06-30 18:54:34,796 [config.py:log_config()] -  ===== Config File =====
[main]
show_cursor = True

[printer Elegoo Neptune 3 Pro]
moonraker_host = 10.0.0.101
moonraker_port = 7125
=======================
2024-06-30 18:54:34,798 [config.py:__init__()] - ====== Saved Def ======

=======================
2024-06-30 18:54:34,798 [config.py:__init__()] - Configured printers: [
  {
    "Elegoo Neptune 3 Pro": {
      "moonraker_host": "10.0.0.101",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2024-06-30 18:54:34,870 [config.py:create_translations()] - Selected lang: system_lang OS lang: en_GB
2024-06-30 18:54:34,871 [config.py:install_language()] - Using lang en
2024-06-30 18:54:34,881 [screen.py:__init__()] - Monitors: 1 using number: 0
2024-06-30 18:54:34,882 [screen.py:__init__()] - Wayland: False Display name: :0
2024-06-30 18:54:34,884 [screen.py:__init__()] - Screen resolution: 1024x768
2024-06-30 18:54:35,070 [KlippyGtk.py:__init__()] - Font size: 25.6 (medium)
2024-06-30 18:54:35,250 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 3600
2024-06-30 18:54:35,258 [screen.py:set_screenblanking_timeout()] - Using DPMS
2024-06-30 18:54:35,267 [screen.py:initial_connection()] - Default printer: None
2024-06-30 18:54:35,291 [screen.py:connect_printer()] - Connecting to printer: Elegoo Neptune 3 Pro
2024-06-30 18:54:35,291 [screen.py:_remove_all_panels()] - Removing all panels
2024-06-30 18:54:35,311 [screen.py:_load_panel()] - Loading panel: splash_screen
2024-06-30 18:54:35,362 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
[... 450 lines cut ...]
]
2024-06-30 19:05:45,508 [config.py:create_translations()] - Selected lang: system_lang OS lang: en_GB
2024-06-30 19:05:45,508 [config.py:install_language()] - Using lang en
2024-06-30 19:05:45,521 [screen.py:__init__()] - Monitors: 1 using number: 0
2024-06-30 19:05:45,523 [screen.py:__init__()] - Wayland: False Display name: :0
2024-06-30 19:05:45,525 [screen.py:__init__()] - Screen resolution: 1024x768
2024-06-30 19:05:45,712 [KlippyGtk.py:__init__()] - Font size: 25.6 (medium)
2024-06-30 19:05:46,202 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 3600
2024-06-30 19:05:46,211 [screen.py:set_screenblanking_timeout()] - Using DPMS
2024-06-30 19:05:46,218 [screen.py:initial_connection()] - Default printer: None
2024-06-30 19:05:46,241 [screen.py:connect_printer()] - Connecting to printer: Elegoo Neptune 3 Pro
2024-06-30 19:05:46,242 [screen.py:_remove_all_panels()] - Removing all panels
2024-06-30 19:05:46,262 [screen.py:_load_panel()] - Loading panel: splash_screen
2024-06-30 19:05:46,312 [screen.py:attach_panel()] - Current panel hierarchy: splash_screen
2024-06-30 19:05:46,337 [KlippyRest.py:_do_request()] - host=10.0.0.101, port=7125>>Max retries exceeded with url>>Failed to establish a new connection>>[Errno 101] Network is unreachable
2024-06-30 19:05:46,338 [screen.py:connect_to_moonraker()] - Cannot get server info
2024-06-30 19:05:50,336 [KlippyRest.py:_do_request()] - host=10.0.0.101, port=7125>>Max retries exceeded with url>>Failed to establish a new connection>>[Errno 111] Connection refused
2024-06-30 19:05:50,336 [screen.py:connect_to_moonraker()] - Cannot get server info
2024-06-30 19:05:54,340 [KlippyRest.py:_do_request()] - host=10.0.0.101, port=7125>>Max retries exceeded with url>>Failed to establish a new connection>>[Errno 111] Connection refused
2024-06-30 19:05:54,341 [screen.py:connect_to_moonraker()] - Cannot get server info
2024-06-30 19:05:58,342 [KlippyRest.py:_do_request()] - host=10.0.0.101, port=7125>>Max retries exceeded with url>>Failed to establish a new connection>>[Errno 111] Connection refused
2024-06-30 19:05:58,343 [screen.py:connect_to_moonraker()] - Cannot get server info
2024-06-30 19:06:02,341 [KlippyRest.py:_do_request()] - host=10.0.0.101, port=7125>>Max retries exceeded with url>>Failed to establish a new connection>>[Errno 111] Connection refused
2024-06-30 19:06:02,341 [screen.py:connect_to_moonraker()] - Cannot get server info
2024-06-30 19:06:06,330 [screen.py:connect_to_moonraker()] - Stopping Retries

trebory6, 2024-07-02:
So after reinstalling and testing on another SD card, I realized that re-installing NetworkManager from the Klipperscreen install screen is what caused the issue.

I ended up backing up all my files and doing a complete reinstall of Moonraker, and when I reinstalled KlipperScreen I made sure to select "no" when it asks if you want to install NetworkManager. Works flawlessly after that.

So I didn't find a way to fix it after the fact, just how to prevent this issue when installing.

closed 2024-07-02 by trebory6

SurfSound, 2024-08-31:
Hi,
I wouldn't consider this completed.
This is not fixed, it is just a workaround.

I have the same issue on the Creality Sonic Pad.
Yesterday I installed "Full Klipper" on it using this: https://github.com/qwazwsx/Advanced-SonicPad
After this I can see and access every WIFI without a problem.
But after the update either through KIAUH or through mainsail when I want to access the network page the message "unexpected error: scanning not allowed while unavailable" occured and there are no WIFI anymore.

I can still connect to my wifi via nmtui but I can not manage anything via the display.
A reinstall of the network-manager did not help.

u0xpsec, 2024-09-01:
> Hi, I wouldn't consider this completed. This is not fixed, it is just a workaround.
>
> I have the same issue on the Creality Sonic Pad. Yesterday I installed "Full Klipper" on it using this: [...]

same thing happened to me today. everything was working fine now I cant scan for networks and klipperscreen keeps showing the same error over and over.

apirkle, 2024-11-13:
Landed here after running into this issue. A quick search shows that Network Manager breaking all WiFi connections is a widespread issue with KlipperScreen installs. Recommend that the maintainers address this.

caiowilson, 2024-12-13:
You can reinstate dhcpd inside `raspi-config` after that you may remove, then install klipperscreen and then purge network-manager.

Barry4u, 2025-01-03:
Installed original Klipper (on FLsun V400), wanted the new features (mostly very happy)
The only issue I have is (like above) when I boot up the wifi only works sometimes. It I boot again it may or may not work.
I have 3 computers in different buildings that need to connect to this printer for configurations & monitoring.
I do not have much time and am very limited in Klipper knowledge. I did manage over 3 days to install, calibrate and get really good prints.
This network manager issue is a real pain, and I am too stubborn to go back to FLsun's version.

The only thing I can think of is to ask for  HELP!

Thank you.