KlipperScreen/KlipperScreen #1423: [BUG] Mistranslation (turkish) on restart system/shutdown system prompt

Dumpflam, 2024-07-04:
### What happened?

i got a moonraker error, fixed it, but when i hit "Sistem yeniden başlat" (restart system) inoitced that it said iptal (cancel) and then "sunucu" (server) . shouldnt that be "yes"? (Evet).... ALSO it should be "sistemi yeniden başlat" (you forgot the "i" after sistem)

### What did you expect to happen instead?

.

### How to reproduce this bug?

.

### Log output

[logs-20240704-211104.zip](https://github.com/user-attachments/files/16102001/logs-20240704-211104.zip)

alfrix, 2024-07-06:
Translations are handled in Weblate, https://hosted.weblate.org/projects/klipperscreen/klipperscreen/
I don't know the language, maybe @technical50 can answer

technical50, 2024-07-06:
Sorunun ne olduğunu bende anlayamadım önce Türkçe sonra İngilizce yazmaya çalış lütfen.

I didn't understand problem also he mixed too much two languages. Please write ur issue first Turkish and try to use English also.

technical50, 2024-07-06:
Home > Ana Ekran Olarak Çeviriliydi (Ev demek yanlış Olduğu için) "Ana konum" Olarak değiştirildi Anlamsız kalabiliyor.

System restart kısımı düzeltildi "Sistemi yeniden başlat"

Host = Sunucu olarak çevirildi ve öyle kalmalı.

Çeviride olan bağlara müdahele edemiyorum. Sunucu Kapa dediğinde gelen bağlar benim elimde değil.

technical50, 2024-07-08:
@alfrix it not bug just suggestions.
You can change it tittle.

Dumpflam, 2024-07-09:
@alfrix yes "host" is not a mistranslation but rather it is in the wrong place. @technical50 
 selim sağol düzelttiğin için kolay gelsin

technical50, 2024-07-09:
> @alfrix yes "host" is not a mistranslation but rather it is in the wrong place. @technical50 selim sağol düzelttiğin için kolay gelsin

Rica ederim sana zahmet konuyu kapatabilir misin.
Thank you to. Please put this Closed title.

alfrix, 2024-07-14:
> > @alfrix yes "host" is not a mistranslation but rather it is in the wrong place. @technical50 selim sağol düzelttiğin için kolay gelsin
> 
> Rica ederim sana zahmet konuyu kapatabilir misin. Thank you to. Please put this Closed title.

Ok, Closing as resolved

closed 2024-07-14 by alfrix