KlipperScreen/KlipperScreen #1428: [BUG] <title>Clicking on the network button shows that there is no sdbus_block module

sybf, 2024-07-09:
### What happened?

Clicking on the network button shows that there is no sdbus_block module

### What did you expect to happen instead?

Displays network management

### How to reproduce this bug?

I don't know how it came to be
[image]

### Additional information:

[image]
[image]

### Log output

[KlipperScreen.log](https://github.com/user-attachments/files/16135713/KlipperScreen.log)

OnAdditive, 2024-07-12:
I'm also having this same bug, updated to KlipperScreen v0.4.3-23-gf9392e67 and suddenly can't access the network menu

sybf, 2024-07-13:
The problem has been solved, uninstall and reinstall it

> 我也有同样的错误，更新到 KlipperScreen v0.4.3-23-gf9392e67，突然无法访问网络菜单

sybf, 2024-07-13:
The problem has been solved, uninstall and reinstall it

closed 2024-07-13 by sybf

Khaos66, 2024-08-13:
@sybf When will you push a new release?
I use Yumi PAD and it checks for new releases to update KlipperScreen

sybf, 2024-08-13:
If the network panel cannot be displayed after updating the klipperscreen, please replace the network environment and uninstall and reinstall the klipperscreen