KlipperScreen/KlipperScreen #1434: temperature too low to extrude

tome9111991, 2024-07-17:
### Requested feature

After one of the latest updates, I get the message "temperature too low to extrude." 
This is quite annoying when I want to execute my load or unload macro. I have written my macros in such a way that the temperatures are set within the macro itself, but unfortunately, this no longer works after the update. 
The macro aborts, and I get the message "temperature too low to extrude." 

Is there a way to deactivate this feature?

tyllurius, 2024-07-27:
Exakt same here. Some new extruder temperature check cause my load/unload macros not to work anymore for same reasons. Same macros started from UI still work as expected.

yzheka, 2024-07-30:
Same. Please revert as it was before.

commit 2c3e0f6c (2024-08-01): refactor(extrude): only open temp panel after sending the command and checking failed

this should allow people to handle the situation without creating an extra setting
close #1434
closed 2024-08-01 by alfrix