KlipperScreen/KlipperScreen #1438: Icon on Main page too large

Thomsen1982, 2024-07-21:
### What happened?

On the right side on the main page I have 8 icons. In the past these icons match the display size without a scroll bar. But something’s is changed.
Is it possible to fix it or is it a config problem?

### What did you expect to happen instead?

Correct the size that the icons are placed without scroll bar

### How to reproduce this bug?

It is every time available.

### Log output

I think a log file is not needed in this case

Thomsen1982, 2024-07-21:
Oh I forgot the screen shot :-)
[image]

alfrix, 2024-07-21:
Not a bug, you have 8 menu entries, they do not fit so there is a scrollbar, i guess you probably copied and pasted all the menu entries from the default config to klipperscreen.conf, which the documentation says multiple times not to do, my recommendation is to remove all the entries that are not custom to your setup, which will be probably all of them

closed 2024-07-21 by alfrix