KlipperScreen/KlipperScreen #1439: [BUG] Layer X out of Y

roykrikke, 2024-07-21:
### What happened?

In Fluidd I can see Layer X out of Y
[image]

In KlipperScreen I can't see this anymore

### What did you expect to happen instead?

Layer X out of Y

### How to reproduce this bug?

Open de screen where Layer X out of Y is normally shown. 

### Additional information:

KlipperScreen: v0.4.3-40

### Log output

[logging.txt](https://github.com/user-attachments/files/16324625/logging.txt)

alfrix, 2024-07-21:
not a bug, klipperscreen will not calculate and guess the layer anymore, provide the correct info:

https://klipperscreen.github.io/KlipperScreen/Quicktips/#accurate-layer-progress-in-the-secondary-screen-of-the-printing-panel

related: https://github.com/KlipperScreen/KlipperScreen/commit/b58471c5057590c5977594882dba2045a173933b

closed 2024-07-21 by alfrix

roykrikke, 2024-07-21:
> not a bug, klipperscreen will not calculate and guess the layer anymore, provide the correct info:
> 
> https://klipperscreen.github.io/KlipperScreen/Quicktips/#accurate-layer-progress-in-the-secondary-screen-of-the-printing-panel
> 
> related: [b58471c](https://github.com/KlipperScreen/KlipperScreen/commit/b58471c5057590c5977594882dba2045a173933b)

Thanks for the quick response!