KlipperScreen/KlipperScreen #1442: Arabic Translation Not Applying in KlipperScreen

haqbany, 2024-07-27:
### What happened?

I have completed the Arabic translation for KlipperScreen on Weblate ([link to translation](https://hosted.weblate.org/projects/klipperscreen/klipperscreen/ar/)). After updating KlipperScreen and selecting Arabic from the language options, no visible changes occur, and the interface remains in the default language.

### What did you expect to happen instead?

The interface should change to Arabic when the Arabic language is selected in the settings.

### How to reproduce this bug?

1. Complete the Arabic translation on Weblate.
2. Update KlipperScreen to the latest version.
3. Select Arabic from the language settings in KlipperScreen.
4. Observe that the interface does not change to Arabic.

### Additional information:

- Completed Arabic translation: [link to translation](https://hosted.weblate.org/projects/klipperscreen/klipperscreen/ar/)
- Verified Arabic language files located at `/home/biqu/KlipperScreen/ks_includes/locales/ar/LC_MESSAGES`

### Log output

[klipper,ar.zip](https://github.com/user-attachments/files/16401212/klipper.ar.zip)

alfrix, 2024-08-01:
not a bug, the translation has not been fully merged yet (#1440)

haqbany, 2024-08-01:
Thank you for the repose , any thing I should do from my end ? More translation?

closed 2024-08-01 by alfrix