KlipperScreen/KlipperScreen #1447: [BUG] installation failed over python version

uyoyo117, 2024-08-06:
### What happened?

the installation failed

### What did you expect to happen instead?

the installation to not fail over false version 

### How to reproduce this bug?

Install python using apt-get, then install python3.8 and update the binery location `echo "alias python=/usr/local/bin/python3.8" >> ~/.bashrc`
and restart

### Log output

###### Cloning KlipperScreen from https://github.com/jordanruthe/KlipperScreen.git ...
Cloning into '/home/pi/KlipperScreen'...
remote: Enumerating objects: 16470, done.
remote: Counting objects: 100% (2198/2198), done.
remote: Compressing objects: 100% (790/790), done.
remote: Total 16470 (delta 1315), reused 1982 (delta 1182), pack-reused 14272
Receiving objects: 100% (16470/16470), 14.96 MiB | 3.68 MiB/s, done.
Resolving deltas: 100% (10772/10772), done.

###### Installing KlipperScreen ...
Checking Python version > 3,8
Python 3.7.3
Not supported

#=======================================================#
 KlipperScreen installation failed!
#=======================================================#

pi@fluiddpi:~ $ python3 --version
Python 3.8.0

alfrix, 2024-08-06:
not a bug, the installer will check the system version and refuse to continue if the version is not correct.

you seem to be doing an alias to python instead of python3, that could be part of your issue, the alias should be done to python3 and not python and the bashrc should be sourced before the install

that being said i'm going to close this. fluiddpi is deprecated, debian buster is EOL, your entire system is not supported anymore and you will encounter further issues if you keep trying to update.
if you do not wish to update your system then do: `git reset --hard v0.4.1` and remove KS from the update manager

closed 2024-08-06 by alfrix

uyoyo117, 2024-08-06:
Do check my end of the log, I have done an alias to both python and python3 and sourced it in the end. 
Calling python3 -V outside the installer gets 3.8..... 
I may look like someone who didn't try anything before coming and complain, but I have already attempted everything I could find, I even check the condition used in the installer script and outside the script it returns the correct version 

`check_requirements()
{
    VERSION="3,8"
    echo_text "Checking Python version > "$VERSION
    python3 --version
    if ! python3 -c 'import sys; exit(1) if sys.version_info <= ('$VERSION') else exit(0)'; then
        echo_error 'Not supported'
        exit 1
    fi
}`