KlipperScreen/KlipperScreen #1465: [FR] Add Turn on Printer to the main screen instead of connection error if there is a configured power switch

Online122228, 2024-09-05:
### Requested feature

instead of showing a warning that the mcu cannot be found, show a switch to turn the printer on

### Is your feature request related to a problem?

you always have to select settings then on and then you can turn on the printer, so three clicks what could be one

### Describe alternatives you have considered

1. turning the printer on over Mainsail
2. turning the printer on over the Wi-Fi plug i use
3. turning the printer on with three clicks instead of one
4. leaving the printer always on
5. leaving the printer always off and never use it anymore

stijnb1234, 2024-09-13:
+1, would be a great feature. Like Mainsail does, show that it's turned off and add a button to switch it on again.

mjfsch, 2024-09-16:
See here https://github.com/KlipperScreen/KlipperScreen/issues/1241#issuecomment-1900205587

Online122228, 2024-09-17:
when i inserted this line into my klipperscreen.conf i got an error, i think it was something like a newline error even if i exactly copy pasted it. i add my moonraker.conf for reference
[moonraker.conf.txt](https://github.com/user-attachments/files/17034987/moonraker.conf.txt)

mjfsch, 2024-09-20:
my klipperscreen.conf reads:

[printer printer]
power_devices: printer
#~# --- Do not edit below this line. This section is auto generated --- #~#

Online122228, 2024-09-20:
So it is split in two lines?

closed 2024-09-20 by Online122228