KlipperScreen/KlipperScreen #1472: [BUG] <title>The wifi function is not available

luckluol, 2024-09-28:
### What happened?

After using the new wifi function, the background will report an error when connecting to the new wifi.

### What did you expect to happen instead?

I do not know whether my dependency library is not correctly added or what, can you indicate how to use the new wifi function in the md file，For example, what libraries and files should be added to support them

### How to reproduce this bug?

sudo apt update
sudo apt install network-manager
sudo apt-get install dbus		
pip install sdbus
pip install --only-binary ':all:' sdbus-networkmanager	

After using the above library, the following bugs have always appeared when using the wifi function

### Log output

Starting WiFi Association
2024-09-28 08:58:44,865 [functions.py:logging_exception_handler()] - Uncaught exception <class 'sdbus.dbus_exceptions.DbusUnknownMethodError'>: No such interface “org.freedesktop.DBus.Properties” on object at path /org/freedesktop/NetworkManager/AccessPoint/47
  File "/home/pi/AB/B/KlipperScreen/panels/network.py", line 210, in add_new_network
    self.connect_network(widget, ssid, showadd=False)

  File "/home/pi/AB/B/KlipperScreen/panels/network.py", line 255, in connect_network
    self.reload_networks()

  File "/home/pi/AB/B/KlipperScreen/panels/network.py", line 399, in reload_networks
    self.load_networks()

  File "/home/pi/AB/B/KlipperScreen/panels/network.py", line 112, in load_networks
    for net in self.sdbus_nm.get_networks():

  File "/home/pi/AB/B/KlipperScreen/ks_includes/sdbus_nm.py", line 178, in get_networks
    networks.extend(

  File "/home/pi/AB/B/KlipperScreen/ks_includes/sdbus_nm.py", line 187, in <genexpr>
    "signal_level": ap.strength,

  File "/home/pi/.local/lib/python3.9/site-packages/sdbus/dbus_proxy_sync_property.py", line 82, in __get__
    reply_message = obj._dbus.attached_bus.call(new_call_message)
NoneType: None

FGmanu, 2024-09-28:
Hello,

I'm having the same problem with a Yumi SmartPad running Klipper since the last updates. After several tests, it seems that the problem comes from the klipperscreen update. As soon as I go from version v0.4.0-35 to v0.4.4-17 the problems appear and the wifi doesn't work anymore with the same error message. 

Is there a solution?

luckluol, 2024-09-29:
The problem seems to be that klipperscreen's new wifi function occasionally fails to obtain the ssid when obtaining the wifi list.
I am currently catching exception handling, but I do not know the specific problem, caught multiple code parts of the exception, now can be used.
If you have a better solution, you can also let me know, thank you

FGmanu, 2024-10-01:
Hello,

Thank you for your feedback.

For the moment, the only solution I've found is to reinstall my Yumi SmartPad and do all the updates except for KlipperScreen, and for the moment, I have no more problems.

I'll see in a couple of KlipperScreen release updates if I take the risk of doing it.

Best Regards

Stefan333de, 2024-10-09:
Same here...
Want do swap a BTT CB2 to an Raspberry CM4 Module in my 3D Printer.

As soon as i install Klipperscreen the WLAN connection is gone.  
Testet Bullseye lite, Bookworm lite 32 and 64bit.
No matter if i choose "no" or "yes" to Networkmanager install.

regards
Stefan

alfrix, 2024-10-09:
do not install like that, the lastest sdbus is not tested, and you should use the installer. (at least follow the installer steps as close as possible if you know what you are doing)

when using raspberryOS bookworm it should work, the installer still does necessary things so it should be run.

if you are using mainsailOS then i would recommend that you wait for the maintainers to release a bookworm version.

FGmanu, 2024-10-09:
Hello,

In my case, I'm on a Yumi SmartPad with YumiOS.

Regards

Stefan333de, 2024-10-12:
Hello,
i have managed to get a working Bookworm Lite 32bit with up to Date Klipper/Klipperscreen.

The only change i have made is to not set the Wifi credentials in the RPI Imager.

Setting up the System without Wifi and give credentials at last with nmtui and all works.

Before i was unable to do anything with nmcli or nmtui.

regards
Stefan

FGmanu, 2024-10-20:
Hello,

For the moment, I haven't updated. I'm waiting for several revisions to potentially make sure it works again.
I'm tired of having to do a SmartPad reset every time I try to update and it doesn't work.