KlipperScreen/KlipperScreen #1473: [FR] dim screen on klipper's idle_timeout

thijstriemstra, 2024-09-29:
### Requested feature

when klipper's idle_timeout kicks in, I'd like to dim the screen as well instead of relying on the auto-dim timeouts. Maybe it can be done in gcode even?

My `idle_timeout`:

```ini
[idle_timeout]
# Idle time (in seconds) to wait before running the idle timeout G-Code
# commands. The default is 600 seconds.
timeout: 300
# A list of G-Code commands to execute on an idle timeout. See
# docs/Command_Templates.md for G-Code format. The default is to run
# "TURN_OFF_HEATERS" and "M84".
gcode:
    TURN_OFF_HEATERS
    M84
    STATUS_OFF
```

ChaosBlades, 2024-10-13:
KlipperScreen can only turn on and off the screen but can not dim the screen unless https://github.com/KlipperScreen/KlipperScreen/pull/1443 is implemented. Best thing that could happen without that is KlipperScreen turning down the pixel brightness like the difference between Brightness and Backlight settings on a TV but implementing it that way would only really work for OLED screens since the pixels on OLED emit their own light. LCDs would still look bright just be harder to see.

KlipperScreen already turns on the screen when print finishes or there is an error. So implementing a feature to call turning off / on the screen at will via gcode seems like it would be straight forward.

thijstriemstra, 2024-10-13:
yea with dim i meant turn off the screen, whatever is feasible.