KlipperScreen/KlipperScreen #1474: [BUG] <title>Error when showing 4 or more filament sensors in extrude menu

JuanR3D, 2024-09-30:
### What happened?

When having only 2 filament sensors, i can access perfectly to their respective buttons, but when adding some other sensors to my config in my case 4), these don't show correctly.

### What did you expect to happen instead?

These should be shown as when having only 2  filament sensors.

### How to reproduce this bug?

Define at least 4 filament sensors and go to extrude menu in klipperscreen.

### Additional information:

[image]
[image]

### Log output

[KlipperScreen.log](https://github.com/user-attachments/files/17185030/KlipperScreen.log)

alfrix, 2024-10-09:
not a bug, that's by design how it's supposed to look, if more than 3 sensors are present, there is no space for the swtiches, if you add 4 switches you will not be able to read the name of the sensors, so they get streamlined

closed 2024-10-09 by alfrix

JuanR3D, 2024-10-10:
Thanks @alfrix  for the response.
Anyways, if i want to enable the control of these sensors, what should i do, create a macro or panel that shows them?
In the case of the print panel, how can i add a button that calls this new macro? (I have tried to add a button in order to call the macro but i only can add it in the "options panel", like below:

[menu __print]
name: {{ gettext('Print Control') }}

[menu __print printoptions]
name: Opciones de extrusion
icon: home
method: printer.gcode.script
params: {"script":"FILAMENT_OPTIONS_PROMPT"}

Thanks again!

alfrix, 2024-10-11:
at some point a "filament sensors" panel will be created for those with multiple filament sensors that do not fit in the extrude panel, but that would be a future feature

> Thanks @alfrix for the response. Anyways, if i want to enable the control of these sensors, what should i do, create a macro or panel that shows them? In the case of the print panel, how can i add a button that calls this new macro? 
> 
> [menu __print] name: {{ gettext('Print Control') }}
> 
> [menu __print printoptions] name: Opciones de extrusion icon: home method: printer.gcode.script params: {"script":"FILAMENT_OPTIONS_PROMPT"}
> 

the that is the way, you don't need the first entry since that is already in the default menu,
if your intention was to create a submenu, then the structure would be like this

```
[menu __print filaments]
icon: spool
name: Opciones de Filamento

[menu __print filaments prompt] 
name: Opciones de extrusion
icon: extrude
method: printer.gcode.script
params: {"script":"FILAMENT_OPTIONS_PROMPT"}
```

to have them outside of printing you will need an extra set like this:

```
[menu __main filaments]
icon: spool
name: Opciones de Filamento

[menu __main filaments prompt] 
name: Opciones de extrusion
icon: extrude
method: printer.gcode.script
params: {"script":"FILAMENT_OPTIONS_PROMPT"}
```

> I have tried to add a button in order to call the macro but i only can add it in the "options panel"

you cannot put extra buttons directly in the job_status panel, that is reserved and not editable

any macro can be called from the macros panel, just hide those that are not used from the screen