KlipperScreen/KlipperScreen #1477: [BUG] After the last update today, klipperscreen does not start

Creo2005, 2024-10-10:
### What happened?

Preparing metadata (setup.py): finished with status 'error' - in log.

I see it on the screen - klipper login: _
But the screen flickers as if klipperscreen is trying to start.

### What did you expect to happen instead?

everything will be fine

### How to reproduce this bug?

Update klipperscreen

### Additional information:

Debian 12 on Intel Core Duo 1.66 with 12" touch screen.
pip -V
pip 23.0.1 from /usr/lib/python3/dist-packages/pip (python 3.11)

### Log output

the last part of the log:
create mode 100644 styles/material-darker/images/battery-25.svg
create mode 100644 styles/material-darker/images/battery-75.svg
create mode 100644 styles/material-darker/images/battery-charging.svg
create mode 100644 styles/material-darker/images/battery-unknown.svg
create mode 100644 styles/material-light/images/battery-100.svg
create mode 100644 styles/material-light/images/battery-25.svg
create mode 100644 styles/material-light/images/battery-unknown.svg
create mode 100644 styles/z-bolt/images/battery-0.svg
create mode 100644 styles/z-bolt/images/battery-50.svg
create mode 100644 styles/z-bolt/images/battery-75.svg
create mode 100644 styles/z-bolt/images/battery-charging.svg
create mode 100644 styles/z-bolt/images/battery-unknown.svg
Git Repo KlipperScreen: Checking pip version...
Git Repo KlipperScreen: Updating pip from version 23.3.2 to 24.0...
Collecting pip==24.0
Using cached pip-24.0-py3-none-any.whl.metadata (3.6 kB)
Using cached pip-24.0-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
Attempting uninstall: pip
Found existing installation: pip 23.3.2
Uninstalling pip-23.3.2:
Successfully uninstalled pip-23.3.2
Successfully installed pip-24.0
Git Repo KlipperScreen: Updating python packages...
Ignoring python-mpv: markers 'python_version < "3.10"' don't match your environment
Ignoring backports.zoneinfo: markers 'python_version < "3.9"' don't match your environment
Requirement already satisfied: jinja2==3.1.4 in /home/creo/.KlipperScreen-env/lib/python3.11/site-packages (from -r /home/creo/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 1)) (3.1.4)
Collecting requests==2.32.3 (from -r /home/creo/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 2))
Using cached requests-2.32.3-py3-none-any.whl.metadata (4.6 kB)
Collecting sdbus==0.11.1 (from -r /home/creo/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 3))
Using cached sdbus-0.11.1.tar.gz (74 kB)
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'error'
Ignoring python-mpv: markers 'python_version < "3.10"' don't match your environment
Ignoring backports.zoneinfo: markers 'python_version < "3.9"' don't match your environment
Requirement already satisfied: jinja2==3.1.4 in /home/creo/.KlipperScreen-env/lib/python3.11/site-packages (from -r /home/creo/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 1)) (3.1.4)
Collecting requests==2.32.3 (from -r /home/creo/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 2))
Using cached requests-2.32.3-py3-none-any.whl.metadata (4.6 kB)
Collecting sdbus==0.11.1 (from -r /home/creo/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 3))
Using cached sdbus-0.11.1.tar.gz (74 kB)
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'error'
Ignoring python-mpv: markers 'python_version < "3.10"' don't match your environment
Ignoring backports.zoneinfo: markers 'python_version < "3.9"' don't match your environment
Requirement already satisfied: jinja2==3.1.4 in /home/creo/.KlipperScreen-env/lib/python3.11/site-packages (from -r /home/creo/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 1)) (3.1.4)
Collecting requests==2.32.3 (from -r /home/creo/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 2))
Using cached requests-2.32.3-py3-none-any.whl.metadata (4.6 kB)
Collecting sdbus==0.11.1 (from -r /home/creo/KlipperScreen/scripts/KlipperScreen-requirements.txt (line 3))
Using cached sdbus-0.11.1.tar.gz (74 kB)
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'error'
Git Repo KlipperScreen: Restarting service KlipperScreen...

Creo2005, 2024-10-10:
[xorg.log](https://github.com/user-attachments/files/17324634/xorg.log)
[KlipperScreen_journalctl.log](https://github.com/user-attachments/files/17324635/KlipperScreen_journalctl.log)
[KlipperScreen_systemctl.log](https://github.com/user-attachments/files/17324636/KlipperScreen_systemctl.log)

Creo2005, 2024-10-10:
`KlipperScreen.service - KlipperScreen
     Loaded: loaded (/etc/systemd/system/KlipperScreen.service; enabled; preset: enabled)
     Active: active (running) since Thu 2024-10-10 11:10:00 EEST; 2s ago
   Main PID: 2399 (KlipperScreen-s)
      Tasks: 3 (limit: 4685)
     Memory: 604.0K
        CPU: 1.377s
     CGroup: /system.slice/KlipperScreen.service
             ├─2399 /bin/bash /home/creo/KlipperScreen/scripts/KlipperScreen-start.sh
             └─2403 /usr/bin/xinit /home/creo/.KlipperScreen-env/bin/python /home/creo/KlipperScreen/screen.py

Oct 10 11:10:02 klipper KlipperScreen-start.sh[2410]:   File "/home/creo/KlipperScreen/screen.py", line 32, in <module>
Oct 10 11:10:02 klipper KlipperScreen-start.sh[2410]:     from panels.base_panel import BasePanel
Oct 10 11:10:02 klipper KlipperScreen-start.sh[2410]:   File "/home/creo/KlipperScreen/panels/base_panel.py", line 3, in <module>
Oct 10 11:10:02 klipper KlipperScreen-start.sh[2410]:     import psutil
Oct 10 11:10:02 klipper KlipperScreen-start.sh[2410]: ModuleNotFoundError: No module named 'psutil'
Oct 10 11:10:02 klipper KlipperScreen-start.sh[2403]: /usr/bin/xinit: connection to X server lost
Oct 10 11:10:02 klipper KlipperScreen-start.sh[2403]:
Oct 10 11:10:02 klipper KlipperScreen-start.sh[2403]: waiting for X server to shut down
Oct 10 11:10:02 klipper KlipperScreen-start.sh[2404]: error setting MTRR (base = 0x00000000d0000000, size = 0x007b0000, type = 1) Invalid argument (22)
Oct 10 11:10:02 klipper KlipperScreen-start.sh[2404]: (II) Server terminated successfully (0). Closing log file.`

malbul, 2024-10-10:
Based on message from the KlipperScreen.service file, `psutil` module is missed.

It seems like `KlipperScreen-requirements.txt` file has missed this dependency.
Activating `KlipperScreen` venv and installing missed module (`psutil`) solves the problem.

alfrix, 2024-10-13:
not a bug, it looks like a packet drop during download, aka internet failure

closed 2024-10-13 by alfrix

Creo2005, 2024-10-13:
> not a bug, it looks like a packet drop during download, aka internet failure
> 
> install the package manually as malbul said, or follow [the manual update procedure](https://klipperscreen.readthedocs.io/en/latest/Updating/#manual-update) or re-run the installer

This is not true. I had no problems with the Internet. Everything was loaded as it should be.

jmeier5261, 2024-10-27:
@Creo2005 I had a very similar issue. Mine looked to be caused by some dependency mismatch due to my mainsail install  being based on buster which didn't have a supported version for something. gave up and rolled both my buster pi's to a bullseye based image